KlipperScreen/KlipperScreen #313: KlipperScreen does not start properly after installation.

hsl416604093, 2021-10-22:
Hi,Gays!
As the question, after installing KlipperScreen, the display still shows terminal login and flashes once in a while.

What is known so far is：

- no /tmp/KlipperScreen.log exists yet;
- `sudo journalctl -xe -u KlipperScreen` gives
[image]
- /etc/X11/Xwrapper.conf has entry` allowed_users=anybody`
- The Xorg.0.log is
[Xorg.0.log](https://github.com/jordanruthe/KlipperScreen/files/7394065/Xorg.0.log)

Any clue what might be wrong?

alfrix, 2021-12-02:
STM32MP157 ? first time i've heard of it
startx is just a script that executes xinit, if it works you may want to take a look at it (usr/bin/startx) and modify the xinit command of the unit file if necessary (etc/systemd/system/klipperscreen.service)

jordanruthe, 2021-12-04:
If you install the desktop version of raspbian, do you have any issues showing an image?

alfrix, 2022-01-22:
Closing because of inactivity

closed 2022-01-22 by alfrix