KlipperScreen/KlipperScreen #322: Power State isn't synced with Moonraker

nicedevil007, 2021-11-07:
Hey jordanruthe,

just small "thing" (I guess so).

After some time the PowerPlug is turned of by klipper during a setup timeout. On fluidd or mainsail the state of the powerplug is shown correctly then but not on KlipperScreen. Is there a way to fix this?

nicedevil007, 2021-11-07:
sry was wrong, that is an issue from moonraker... it didn't check from time to time the right state of the Powerdevice and tho it is in wrong state after shutdown :(

nicedevil007, 2021-11-07:
Mhhh, no it seems to be an issue on KlipperScreen Site after testing multiple times now:

State on Fluidd/Moonraker:
[image]

And here the state on KlipperScreen:
[image]

hawkeyexp, 2021-11-08:
i can confirm the same with my sonoff-devices which are attached and working.

jordanruthe, 2021-12-02:
Let me know if you still have problems.

closed 2021-12-02 by jordanruthe

hapklaar, 2021-12-18:
@jordanruthe The problem persists. After Fluidd switches the Tasmota device off, KlipperScreen does not pick this up and still shows it as ON. Have to switch it off, then on again to power on the printer from KlipperScreen

commit dd0fd18b (2021-12-02): power: Sync power state on panel load. Fixes #322