KlipperScreen/KlipperScreen #328: KlipperScreen not completing connection on boot

julianschill, 2021-11-15:
When starting the printer, sometimes KlipperScreen cannot connect, because moonraker/klipper is not ready yet. According to the logs it gets an 404 on the printer/info endpoint. But instead of trying again it just stays in this state until you restart klipper.

In this log Klipper was restartet at 16:38:55, 10 minutes after switching on the printer. 

```
2021-11-15 16:28:02,249 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-15 16:28:02,254 [KlippyWebsocket.py:on_error()] - Websocket error: [Errno 111] Connection refused
2021-11-15 16:28:02,254 [KlippyWebsocket.py:on_close()] - Connection already closed
2021-11-15 16:28:02,739 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-15 16:28:02,739 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2021-11-15 16:28:02,742 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-15 16:28:02,751 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 200 46
2021-11-15 16:28:02,758 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2021-11-15 16:28:02,759 [KlippyWebsocket.py:on_open()] - Self.connected = False
2021-11-15 16:28:02,760 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2021-11-15 16:28:02,763 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-15 16:28:02,778 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 404 390
2021-11-15 16:28:02,780 [screen.py:init_printer()] - Unable to get printer info from moonraker
2021-11-15 16:38:55,726 [screen.py:_websocket_callback()] - Received notify_klippy_disconnected
2021-11-15 16:38:59,852 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2021-11-15 16:38:59,853 [printer.py:change_state()] - Adding callback for state: ready
2021-11-15 16:38:59,856 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2021-11-15 16:38:59,865 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-15 16:38:59,878 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 200 352
2021-11-15 16:38:59,881 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks
2021-11-15 16:38:59,889 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-15 16:39:00,072 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks HTTP/1.1" 200 89155
```

A possible solution could be to poll the endpoint until the info can be obtained.

Hardware used: Raspberry Pi 4 on Fluiddpi

Lino77, 2021-11-15:
I also had this problem,
gap
ExecStartPre = / bin / sleep 10 in
/etc/systemd/system/KlipperScreen.service that solved the problem for me

[Unit]
Description=KlipperScreen
After=moonraker.service
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=1
User=pi
WorkingDirectory=/home/pi/KlipperScreen
ExecStart=/usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/Klippe$
ExecStartPre=/bin/sleep 10

[Install]
WantedBy=multi-user.target

screech24, 2021-11-15:
Can confirm I have the same issue after recent Moonraker Updates..

NikosSprocket, 2021-11-16:
From initial power-up of a Voron 2.4 running klipper v0.10.0-124-g5dcc377c and KlipperScreen v0.1.6-129-gf6fcb32 the KlipperScreen fails to connect to Klipper. Moonraker is operating as usual and everything looks good. After a "Klipper Restart" from Moonraker or KlipperScreen KlipperScreen is functional again. From the KlipperScreen.log looks like there is an issue with a Python call/variable
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7549856/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/7549858/klippy.log)
?

jordanruthe, 2021-11-17:
Let me know if the commit I just pushed resolves the connect issues. @NikosSprocket I have still to investigate that error, but this may address it anyways.

NikosSprocket, 2021-11-17:
Thanks @jordanruthe!
I just updated to v0.1.6-135-g2b2e096 and run the following tests and it is working again as before; just taking ~10 sec to connect to Klipper.
1. Use Klipper Reboot - Good!
2. Shutdown Klipper, power-off everything and then power-on - Good!

Oh just and fyi, the only other upgrade I have done was rpi had a package update for the rpi-eeprom. I have note updated to Klipper v0.10.0-130 yet. Still running v0.10.0-124.

Cheers!

jordanruthe, 2021-11-22:
@julianschill Can you confirm if this issue is now fixed for you?

julianschill, 2021-11-24:
Fixed for me on the newest version.

closed 2021-11-24 by jordanruthe

NikosSprocket, 2021-11-24:
Still no issues, and  I have updated OS, and Klipper to the current stable versions.

LYRA3D, 2023-03-28:
Hi, I have the same problem

trebory6, 2024-07-02:
Oddly enough I'm having the same issue as in this thread. 

Klipperscreen boots up and immediately says:

```
Cannot Connect to Moonraker

hots=127.0.0.1, port=7125
Max retries exceeded with url
Failed to establish new connection
[Errno 111] Connection Refused
```

But when I click into and restart Klipperscreen it connects fine. 

Have to do this for every power cycle now.

trebory6, 2024-07-02:
Oddly enough I'm having the same issue as in this thread. 

Klipperscreen boots up and immediately says:

```
Cannot Connect to Moonraker

hots=127.0.0.1, port=7125
Max retries exceeded with url
Failed to establish new connection
[Errno 111] Connection Refused
```

But when I click into and restart Klipperscreen it connects fine. 

Have to do this for every power cycle now. 

[image]

appleimperio, 2024-07-24:
@trebory6 I'm having the same issue. Did you find the solution?

qthegaijin, 2024-07-27:
@trebory6 I too am having this issue now on a fresh install. Any updates?

trebory6, 2024-08-26:
@appleimperio @qthegaijin Unfortunately not. 

It was because of this and a lack of ventilation on my raspberry I just removed it. I might re-visit later when I get a better screen setup for my raspi.

However the issue seems to be that Klipperscreen opens up faster than it takes Klipper to connect to the printer's MCU, and because it attempts to connect several times in a row before Klipper/the MCU is ready, it times out and gives this error.

When I was having this issue I was thinking that writing a script that restarts the KlipperScreen service and is delayed by a couple of minutes after start when the MCU is likely to be connected, or triggered by an event like a successful connection to the MCU, could be a viable solution. 

Sorry I can't be more help, but hope this helps points someone in the right direction.