KlipperScreen/KlipperScreen #330: KlipperScreen not starting up

victorlu0071, 2021-11-17:
It appeared to have no log files.
also my hardware is waveshare 5.5inch OLED screen connecting via HDMI

May i know if klipperscreen use fbturbo since waveshare ask me to # that line in order to rotate to landscape mode.
i tried with /without # that line get the same result. My screen shows switching from console to blank to console to blank constantly 
pi@voron:~ $ journalctl -xe -u KlipperScreen
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit KlipperScreen.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Nov 17 05:43:43 voron systemd[1]: Stopped KlipperScreen.
-- Subject: A stop job for unit KlipperScreen.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit KlipperScreen.service has finished.
--
-- The job identifier is 4504 and the job result is done.
Nov 17 05:43:43 voron systemd[1]: Started KlipperScreen.
-- Subject: A start job for unit KlipperScreen.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit KlipperScreen.service has finished successfully.
--
-- The job identifier is 4504.
Nov 17 05:43:43 voron xinit[2010]: X.Org X Server 1.20.4
Nov 17 05:43:43 voron xinit[2010]: X Protocol Version 11, Revision 0
Nov 17 05:43:43 voron xinit[2010]: Build Operating System: Linux 5.10.17-v8+ aarch64 Raspbian
Nov 17 05:43:43 voron xinit[2010]: Current Operating System: Linux voron 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST
Nov 17 05:43:43 voron xinit[2010]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=
Nov 17 05:43:43 voron xinit[2010]: Build Date: 19 April 2021  03:23:36PM
Nov 17 05:43:43 voron xinit[2010]: xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Nov 17 05:43:43 voron xinit[2010]: Current version of pixman: 0.36.0
Nov 17 05:43:43 voron xinit[2010]:         Before reporting problems, check http://wiki.x.org
Nov 17 05:43:43 voron xinit[2010]:         to make sure that you have the latest version.
Nov 17 05:43:43 voron xinit[2010]: Markers: (--) probed, (**) from config file, (==) default setting,
Nov 17 05:43:43 voron xinit[2010]:         (++) from command line, (!!) notice, (II) informational,
Nov 17 05:43:43 voron xinit[2010]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Nov 17 05:43:43 voron xinit[2010]: (==) Log file: "/var/log/Xorg.0.log", Time: Wed Nov 17 05:43:43 2021
Nov 17 05:43:43 voron xinit[2010]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Nov 17 05:43:43 voron xinit[2010]: (II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
Nov 17 05:43:44 voron xinit[2010]: Traceback (most recent call last):
Nov 17 05:43:44 voron xinit[2010]:   File "/home/pi/KlipperScreen/screen.py", line 4, in <module>
Nov 17 05:43:44 voron xinit[2010]:     import gi
Nov 17 05:43:44 voron xinit[2010]: ModuleNotFoundError: No module named 'gi'
Nov 17 05:43:44 voron xinit[2010]: /usr/bin/xinit: connection to X server lost
Nov 17 05:43:44 voron xinit[2010]: [1B blob data]
Nov 17 05:43:45 voron xinit[2010]: waiting for X server to shut down (II) Server terminated successfully (0). Closing lo
Nov 17 05:43:45 voron systemd[1]: KlipperScreen.service: Succeeded.
-- Subject: Unit succeeded
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit KlipperScreen.service has successfully entered the 'dead' state.
lines 1833-1881/1881 (END)

anythoughts?

victorlu0071, 2021-11-17:
comment dtoverlay=vc4-fkms-v3d on config.txt as requested by waveshare staff for debugging shows another type of error
Nov 17 06:59:00 voron systemd[1]: Started KlipperScreen.
-- Subject: A start job for unit KlipperScreen.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit KlipperScreen.service has finished successfully.
--
-- The job identifier is 8601.
Nov 17 06:59:00 voron xinit[7310]: X.Org X Server 1.20.4
Nov 17 06:59:00 voron xinit[7310]: X Protocol Version 11, Revision 0
Nov 17 06:59:00 voron xinit[7310]: Build Operating System: Linux 5.10.17-v8+ aarch64 Raspbian
Nov 17 06:59:00 voron xinit[7310]: Current Operating System: Linux voron 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST
Nov 17 06:59:00 voron xinit[7310]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=
Nov 17 06:59:00 voron xinit[7310]: Build Date: 19 April 2021  03:23:36PM
Nov 17 06:59:00 voron xinit[7310]: xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Nov 17 06:59:00 voron xinit[7310]: Current version of pixman: 0.36.0
Nov 17 06:59:00 voron xinit[7310]:         Before reporting problems, check http://wiki.x.org
Nov 17 06:59:00 voron xinit[7310]:         to make sure that you have the latest version.
Nov 17 06:59:00 voron xinit[7310]: Markers: (--) probed, (**) from config file, (==) default setting,
Nov 17 06:59:00 voron xinit[7310]:         (++) from command line, (!!) notice, (II) informational,
Nov 17 06:59:00 voron xinit[7310]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Nov 17 06:59:00 voron xinit[7310]: (==) Log file: "/var/log/Xorg.0.log", Time: Wed Nov 17 06:59:00 2021
Nov 17 06:59:00 voron xinit[7310]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Nov 17 06:59:00 voron xinit[7310]: (EE)
Nov 17 06:59:00 voron xinit[7310]: Fatal server error:
Nov 17 06:59:00 voron xinit[7310]: (EE) no screens found(EE)
Nov 17 06:59:00 voron xinit[7310]: (EE)
Nov 17 06:59:00 voron xinit[7310]: Please consult the The X.Org Foundation support
Nov 17 06:59:00 voron xinit[7310]:          at http://wiki.x.org
Nov 17 06:59:00 voron xinit[7310]:  for help.
Nov 17 06:59:00 voron xinit[7310]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional informati
Nov 17 06:59:00 voron xinit[7310]: (EE)
Nov 17 06:59:00 voron xinit[7310]: (EE) Server terminated with error (1). Closing log file.
Nov 17 06:59:15 voron xinit[7310]: /usr/bin/xinit: giving up
Nov 17 06:59:15 voron xinit[7310]: /usr/bin/xinit: unable to connect to X server: Connection refused
Nov 17 06:59:15 voron xinit[7310]: /usr/bin/xinit: server error
Nov 17 06:59:15 voron systemd[1]: KlipperScreen.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit KlipperScreen.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Nov 17 06:59:15 voron systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit KlipperScreen.service has entered the 'failed' state with result 'exit-code'.

victorlu0071, 2021-11-17:
for the sake of indentifying the problem, i attached a short video  
https://youtu.be/AgUBkztaZ6Y

Hundsbuah, 2021-11-19:
Same Problem. Any solution yet?

jordanruthe, 2021-11-22:
@victorlu0071 are you still having issues? Could you update with your journalctl log per the troubleshooting page in the docs? The first error I saw was a problem with the installation, so it would be good to re-run the installation script.

victorlu0071, 2021-11-30:
imma going to try it again after finish rebuilding my PC would thank everyone for their time. I`ll report back when Im done

alfrix, 2022-01-21:
Closing because of inactivity, many things have changed since this issue was posted.

if you want to reinstall KS, delete /home/pi/.KlipperScreen-env prior to the installer.

closed 2022-01-21 by alfrix