KlipperScreen/KlipperScreen #331: Connecting to printer

andytran1, 2021-11-17:
I kept getting this error "Connecting to printer" when KlipperScreen started. It seems like it cannot connect to Moonraker service.
I installed KlipperScreen via Kiah script.
Here are my current setup:
Klipper version: v0.10.0-130
Moonraker version: v0.7.1-152
Fluidd version: v1.16.2
KlipperScreen version: v0.1.6-135

Here are some log entries from `/tmp/KlipperScreen.log`

```
2021-11-17 12:23:44,747 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:44,760 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:45,240 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:45,240 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:45,248 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:45,257 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:45,741 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:45,741 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:45,748 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:45,761 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:46,241 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:46,245 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:46,253 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:46,262 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:46,742 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:46,743 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:46,750 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:46,762 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:47,243 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:47,243 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:47,247 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:47,252 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
2021-11-17 12:23:47,743 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-11-17 12:23:47,744 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2021-11-17 12:23:47,747 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-11-17 12:23:47,754 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 431
```
Current `moonraker.config` file:
```
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: /home/pi/klipper_config
temperature_store_size: 600
gcode_store_size: 1000
log_path: ~/klipper_logs

[authorization]
force_logins: false
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
  127.0.0.1
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
enable_auto_refresh: True

# this enabled fluidd updates
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd
```
I can connect to Fluidd webserver fine, but KlipperScreen doesn't. Any helps are welcome.

jordanruthe, 2021-11-22:
Can you paste the results of `http://fluidd:7125/server/info`? Please replace `fluidd` with your moonraker address.

HonzaDandy, 2022-01-19:
I have the same problem. Here is my http://fluidd:7125/server/info result:
`{"result": {"klippy_connected": true, "klippy_state": "ready", "components": ["application", "websockets", "internal_transport", "database", "file_manager", "klippy_apis", "machine", "data_store", "shell_command", "proc_stats", "job_state", "job_queue", "authorization", "octoprint_compat", "history"], "failed_components": ["update_manager"], "registered_directories": ["config", "logs", "gcodes", "config_examples", "docs"], "warnings": ["[authorization]: Invalid CIDR expression '127.0.0.1/8' in option 'trusted_clients'", "Update Manger KlipperScreen: Unable to locate pip executable", "Update Manger KlipperScreen: Unable to locate pip executable", "Update Manger KlipperScreen: Unable to locate pip executable", "Unparsed config option 'origin: https://github.com/jordanruthe/KlipperScreen.git' detected in section [update_manager KlipperScreen].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'requirements: scripts/KlipperScreen-requirements.txt' detected in section [update_manager KlipperScreen].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'install_script: scripts/KlipperScreen-install.sh' detected in section [update_manager KlipperScreen].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error."], "websocket_count": 1, "moonraker_version": "v0.7.1-300-g5f54a23"}}`

alfrix, 2022-01-21:
OP did not follow through the troubleshooting help, closing because of inactivity

closed 2022-01-21 by alfrix