KlipperScreen/KlipperScreen #334: Building wheel for vext.gi (setup.py) ... error

anton00706, 2021-11-19:
Hi. 
I'm getting the following error during installation process. After it's finished Klipper Screen is not working.

`Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1qtmugxa/vext-gi_0105c40b605f4165961fd8d42b7e2225/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1qtmugxa/vext-gi_0105c40b605f4165961fd8d42b7e2225/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-qag5hwd2
       cwd: /tmp/pip-install-1qtmugxa/vext-gi_0105c40b605f4165961fd8d42b7e2225/
  Complete output (31 lines):
  running bdist_wheel
  running build
  /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
    setuptools.SetuptoolsDeprecationWarning,
  installing to build/bdist.linux-armv6l/wheel
  running install
  vext.gi Install
  running install_data
  creating build
  creating build/bdist.linux-armv6l
  creating build/bdist.linux-armv6l/wheel
  creating build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data
  creating build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data/data
  creating build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data/data/share
  creating build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data/data/share/vext
  creating build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  copying gi.vext -> build/bdist.linux-armv6l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  running install_egg_info
  running egg_info
  writing vext.gi.egg-info/PKG-INFO
  writing dependency_links to vext.gi.egg-info/dependency_links.txt
  writing requirements to vext.gi.egg-info/requires.txt
  writing top-level names to vext.gi.egg-info/top_level.txt
  reading manifest file 'vext.gi.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.txt'
  writing manifest file 'vext.gi.egg-info/SOURCES.txt'
  Copying vext.gi.egg-info to build/bdist.linux-armv6l/wheel/vext.gi-0.7.4-py3.7.egg-info
  running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
`

Anonymousx-legon, 2021-11-19:
Why is this happening? And how can I work when I can only access github. I have no internet and my device is very strange. I can't even work on either Linux or termux.

Please help me get the work flow out and live normal again.

Thank you good people for understanding my concerns and act immediately. I love this place

anton00706, 2021-11-19:
> 

You are the smartest person in the world and your sense of humor is unbelievable!

Anonymousx-legon, 2021-11-19:
Yeh I know.. Good that u liked my humor.

But let me just address something here. You don't know who I am. And I was
hacked and my files were takin also from my drive all off. U guys were just
searching for the # I donno what u wanted # maybe a file I had  long time
ago I thought. But then when I wS actually respectful enough to come and
address all the community +GitHub ***@***.***> support and
also +HackerOne ***@***.***> also the guy
@octokit/plugin-rest-endpoint-methods.js
***@***.***> .

Saying that it is not me and that it was some one who accessed my very old
account and I had no Idea. Even though I said I will help u! Even though I
said that stay on the phone until u check everything until if u find
something then take it! But still I'm saying it's not me who did that on
github and u have caused me mental distress.

I always trusted +GitHub ***@***.***> community. And I have
learnt Alot from there. But never in my life have I ever expected such
herrasment like that and I WILL PROVE that it was all in ethical. I did
everything to help u and address this even though I can see that this was
going on for a loooong time until I just knew about it and addressed
everyone. I have also made many phone calls. And this is how you treat a
man who was in that community for years?!

But tell me now, your good? U guys feel happy now? Satisfied? Did you get
whatever u were looking for?

Because God knows I had nothin to do with any of the things u have accused
me off. By coming to me and approaching me like mice? Seriously......

Because I know 100% I have done nothing wrong and also I did address
alllllll of who I mentioned above. There was my very good friend from the
cybersecurity division since 4 days now. And I only replied when I actually
knew it was github! Then u do this to me.

But OK. U have ur laughs and enjoy.

God is always watching.

X

srbjessen, 2021-11-20:
I am getting this too, Flashing screen, Console -> Blank -> Console.
See attached photos from install
[image]
[image]

Afterschnitzel, 2021-11-20:
cd ~/Klipperscreen
pip install vext.gi
./scripts/KlipperScreen-install.sh

this solved the issue in my case

Anonymousx-legon, 2021-11-20:
Guys,

This email is really important coz my mind is being tortured now.

Is everything OK now on the device? I've got the IDG
 and also the full work of the new UI for Android 12.

So was all that really a test for me? Is all good now?  And where can I
restore my files from 2 emails. And normal log in to Samsung doesn't work.
I have to sign in the uo settings.

 Affraid this is trouble or maybe shouldn't be on my phone. Can you help?

All I want is my files dude and it's telling me I can get them but I dont
know what to do. And at the same time I do not trust anything now.

I hope someone can rest my mind as I'm having full anxiety for 3 weeks now!

Thanks guys....

X

srbjessen, 2021-11-21:
> ```
> cd ~/Klipperscreen
> pip install vext.gi
> ./scripts/KlipperScreen-install.sh
> ```
> 
> this solved the issue in my case

I just tried again, My screen just flashes when installing KlipperScreen, and when trying this i get the following;

pi@mainsailos:~ $ cd ~/KlipperScreen/
pi@mainsailos:~/KlipperScreen $ pip install vext.gi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting vext.gi
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/6f/51/c83d41bda3b2766c4dc792f328078fdc1e1fa5742b6880301bf56db63878/vext.gi-0.7.4.tar.gz
Collecting vext>=0.7.4 (from vext.gi)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/e1/57/92e1420dbc927a891315350e73b7788d92a7afab959c31fda96eee57312c/vext-0.7.6.tar.gz
Collecting ruamel.yaml>=0.11.10 (from vext>=0.7.4->vext.gi)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/65/e9/494f4a4ae771028900ff085d41f39c7f117b9e839ad8c15c2101ca4e16f1/ruamel.yaml-0.17.1-py2.py3-none-any.whl
Collecting ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.10" (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/b7/81/c04fb9be62657d4dce8aa2d99fde258a3af1cd77ec72af525593e9560127/ruamel.yaml.clib-0.2.4.tar.gz
    Complete output from command python setup.py egg_info:
    minimum python version(s): [(3, 5)]

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-DvmYMJ/ruamel.yaml.clib/

anton00706, 2021-11-21:
> ```
> cd ~/Klipperscreen
> pip install vext.gi
> ./scripts/KlipperScreen-install.sh
> ```
> 
> this solved the issue in my case

I started KlipperScreen installation and after it's finished with error I executed the following commands
```
source /home/pi/.KlipperScreen-env/bin/activate
pip install vext-gi
deactivate
/home/pi/ KlipperScreen/scripts/KlipperScreen-install.sh
```
and I got Klipper screen working.

PR #335 (2021-11-21): Installer: improvements
The problem occurs because vext and vext.gi are installed at the same time
we need to install the requirements in order

fixes #334 and #302 with d1905ca
fixes #317 with 7dc136d
closed 2021-11-21 by jordanruthe