KlipperScreen/KlipperScreen #349: KlipperScreen not start - undefined symbol: shadowUpdatePackedWeak

radaiko, 2021-12-02:
KlipperScreen is not starting.

> sudo systemctl status KlipperScreen
KlipperScreen.service - KlipperScreen
   Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-12-02 14:56:51 CET; 19s ago
 Main PID: 948 (xinit)
    Tasks: 2 (limit: 2138)
   CGroup: /system.slice/KlipperScreen.service
           └─948 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py

> Dec` 02 14:56:51 BedSlinger xinit[948]: Current version of pixman: 0.36.0
Dec 02 14:56:51 BedSlinger xinit[948]:         Before reporting problems, check http://wiki.x.org
Dec 02 14:56:51 BedSlinger xinit[948]:         to make sure that you have the latest version.
Dec 02 14:56:51 BedSlinger xinit[948]: Markers: (--) probed, (**) from config file, (==) default setting,
Dec 02 14:56:51 BedSlinger xinit[948]:         (++) from command line, (!!) notice, (II) informational,
Dec 02 14:56:51 BedSlinger xinit[948]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Dec 02 14:56:51 BedSlinger xinit[948]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Dec  2 14:56:51 2021
Dec 02 14:56:51 BedSlinger xinit[948]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Dec 02 14:56:52 BedSlinger xinit[948]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.63-v8+
Dec 02 14:56:52 BedSlinger xinit[948]: /usr/lib/xorg/Xorg: symbol lookup error: /usr/lib/xorg/modules/drivers/fbturbo_drv.so: undefined symbol: shadowUpdatePackedWeak

Hardware: Raspberry Pi 4 with an OSOYOO 5 Inch DSI Touch Sceen LCD Display 800 x 480

Screen worked beginning of 2021 with Octoscreen so shouldn't be a hardware problem.

alfrix, 2021-12-02:
Try
`sudo apt purge xserver-xorg-video-fbturbo`

radaiko, 2021-12-02:
Thanks that did worked.

closed 2021-12-02 by radaiko

alfrix, 2021-12-02:
Blacklist the package or it will install again when you update

radaiko, 2021-12-02:
Just to get the full information here for others.

To blacklist this package add 

> Package: xserver-xorg-video-fbturbo
Pin: origin ""
Pin-Priority: -1

to the file /etc/apt/preferences

julianschill, 2022-10-16:
I had the same issue when running KlipperScreen on 64bit Linux on a Raspberry Pi 4. Apparently KlipperScreen installs this package with the installation script. Would it make sense to remove that package from the installation script? For what is it needed?