KlipperScreen/KlipperScreen #354: I have two issues pycairo and xserver-xorg-video-fbturbo

bhast2, 2021-12-06:
This is the errors I get when I install How can I fix this
First fbturbo error (this is all it says)
`Unable to locate package xserver-xorg-video-fbturbo`

Then it won't install pycairo

```
bhast2@orangepizero2:~/KlipperScreen$ ./scripts/KlipperScreen-install.sh
Update package data
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/debian buster InRelease
Hit:2 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates InRelease
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates InRelease
Reading package lists... Done
Checking for broken packages...
No broken packages
Installing KlipperScreen dependencies
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package xserver-xorg-video-fbturbo
Creating virtual environment
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/bhast2/.KlipperScreen-env/bin/python3
Also creating executable in /home/bhast2/.KlipperScreen-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...Unable to locate package xserver-xorg-video-fbturbodone.
Requirement already satisfied: pip in /home/bhast2/.KlipperScreen-env/lib/python3.7/site-packages (21.3.1)
Collecting humanize==3.13.1
  Using cached humanize-3.13.1-py3-none-any.whl (98 kB)
Collecting importlib-metadata
  Using cached importlib_metadata-4.8.2-py3-none-any.whl (17 kB)
Collecting typing-extensions>=3.6.4
  Using cached typing_extensions-4.0.1-py3-none-any.whl (22 kB)
Collecting zipp>=0.5
  Using cached zipp-3.6.0-py3-none-any.whl (5.3 kB)
Installing collected packages: zipp, typing-extensions, importlib-metadata, humanize
Successfully installed humanize-3.13.1 importlib-metadata-4.8.2 typing-extensions-4.0.1 zipp-3.6.0
Collecting jinja2==3.0.3
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Collecting MarkupSafe>=2.0
  Using cached MarkupSafe-2.0.1-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (26 kB)
Installing collected packages: MarkupSafe, jinja2
Successfully installed MarkupSafe-2.0.1 jinja2-3.0.3
Collecting matplotlib==3.5.0
  Using cached matplotlib-3.5.0-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (11.6 MB)
Collecting pyparsing>=2.2.1
  Using cached pyparsing-3.0.6-py3-none-any.whl (97 kB)
Collecting python-dateutil>=2.7
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting cycler>=0.10
  Using cached cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting setuptools-scm>=4
  Using cached setuptools_scm-6.3.2-py3-none-any.whl (33 kB)
Collecting fonttools>=4.22.0
  Using cached fonttools-4.28.3-py3-none-any.whl (884 kB)
Collecting pillow>=6.2.0
  Using cached Pillow-8.4.0-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (3.0 MB)
Collecting kiwisolver>=1.0.1
  Using cached kiwisolver-1.3.2-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (1.4 MB)
Collecting packaging>=20.0
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting numpy>=1.17
  Using cached numpy-1.21.4-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (13.0 MB)
Collecting six>=1.5
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Requirement already satisfied: setuptools in /home/bhast2/.KlipperScreen-env/lib/python3.7/site-packages (from setuptools-scm>=4->matplotlib==3.5.0) (59.5.0)
Collecting tomli>=1.0.0
  Using cached tomli-1.2.2-py3-none-any.whl (12 kB)
Installing collected packages: pyparsing, tomli, six, packaging, setuptools-scm, python-dateutil, pillow, numpy, kiwisolver, fonttools, cycler, matplotlib
Successfully installed cycler-0.11.0 fonttools-4.28.3 kiwisolver-1.3.2 matplotlib-3.5.0 numpy-1.21.4 packaging-21.3 pillow-8.4.0 pyparsing-3.0.6 python-dateutil-2.8.2 setuptools-scm-6.3.2 six-1.16.0 tomli-1.2.2
Collecting netifaces==0.11.0
  Using cached netifaces-0.11.0-cp37-cp37m-linux_aarch64.whl
Installing collected packages: netifaces
Successfully installed netifaces-0.11.0
Collecting requests==2.26.0
  Using cached requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting charset-normalizer~=2.0.0
  Using cached charset_normalizer-2.0.9-py3-none-any.whl (39 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2021.10.8-py2.py3-none-any.whl (149 kB)
Collecting idna<4,>=2.5
  Using cached idna-3.3-py3-none-any.whl (61 kB)
Collecting urllib3<1.27,>=1.21.1
  Using cached urllib3-1.26.7-py2.py3-none-any.whl (138 kB)
Installing collected packages: urllib3, idna, charset-normalizer, certifi, requests
Successfully installed certifi-2021.10.8 charset-normalizer-2.0.9 idna-3.3 requests-2.26.0 urllib3-1.26.7
Collecting websocket-client==1.2.1
  Using cached websocket_client-1.2.1-py2.py3-none-any.whl (52 kB)
Installing collected packages: websocket-client
Successfully installed websocket-client-1.2.1
Collecting pycairo==1.20.1
  Using cached pycairo-1.20.1.tar.gz (344 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pycairo
  Building wheel for pycairo (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/bhast2/.KlipperScreen-env/bin/python3 /home/bhast2/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmprnw9ewio
       cwd: /tmp/pip-install-4ckque0r/pycairo_d8c38a76a0ce414c8bf062adc9ed3e68
  Complete output (15 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.7
  creating build/lib.linux-aarch64-3.7/cairo
  copying cairo/__init__.py -> build/lib.linux-aarch64-3.7/cairo
  copying cairo/__init__.pyi -> build/lib.linux-aarch64-3.7/cairo
  copying cairo/py.typed -> build/lib.linux-aarch64-3.7/cairo
  running build_ext
  Package cairo was not found in the pkg-config search path.
  Perhaps you should add the directory containing `cairo.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'cairo' found
  Command '['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for pycairo
Failed to build pycairo
ERROR: Could not build wheels for pycairo, which is required to install pyproject.toml-based projects
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.
bhast2@orangepizero2:~/KlipperScreen$ Unable to locate package xserver-xorg-vide
```

alfrix, 2021-12-06:
fbturbo probably doesn't apply to you platform, usually not a problem

for cairo the only packages needed are handled by the installer:
`sudo apt install libcairo2-dev pkg-config python3-dev`
maybe installing it at system level would help try:
`sudo apt install python-gi-cairo`

orobardet, 2021-12-27:
I add a similar isssue yesterday installing KlipperScreen in a Ubuntu Focal 20.04 for development.

I had to install two more packages (maybe they need to be added in the install script?) :
```
sudo apt install libpango1.0-dev libgirepository1.0-dev 
```
This fixed my issue.

alfrix, 2022-01-27:
Closing because of inactivity, and the fact that the op never replied

closed 2022-01-27 by alfrix