KlipperScreen/KlipperScreen #360: Unable to load main_menu

th3fallen, 2021-12-09:
```2021-12-09 08:34:03,919 [screen.py:_load_panel()] - Loading panel: main_menu
2021-12-09 08:34:03,919 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/main_menu.py
2021-12-09 08:34:04,004 [main_menu.py:add_device()] - Adding device: heater_bed
2021-12-09 08:34:04,005 [KlippyGtk.py:get_temp_color()] - Color list {'extruder': {'colors': ['DC322F', 'B58900', 'CB4B16', 'AA1F1D', '973911'], 'state': 0}, 'bed': {'colors': ['268BD2'], 'state': 0}, 'fan': {'colors': ['859900', '2AA198', '637300', '1F7A72'], 'state': 0}, 'sensor': {'colors': ['D33682', '6C71C4', 'C06CC4', '6D26D1', '2A34A1'], 'state': 0}}
2021-12-09 08:34:04,006 [KlippyGtk.py:get_temp_color()] - Assigning color: bed [0.14901960784313725, 0.5450980392156862, 0.8235294117647058] 268BD2
2021-12-09 08:34:04,014 [main_menu.py:add_device()] - Adding device: frame_expansion_compensation
2021-12-09 08:34:04,015 [KlippyGtk.py:get_temp_color()] - Color list {'extruder': {'colors': ['DC322F', 'B58900', 'CB4B16', 'AA1F1D', '973911'], 'state': 0}, 'bed': {'colors': ['268BD2'], 'state': 1}, 'fan': {'colors': ['859900', '2AA198', '637300', '1F7A72'], 'state': 0}, 'sensor': {'colors': ['D33682', '6C71C4', 'C06CC4', '6D26D1', '2A34A1'], 'state': 0}}
2021-12-09 08:34:04,015 [KlippyGtk.py:get_temp_color()] - Assigning color: sensor [0.8274509803921568, 0.21176470588235294, 0.5098039215686274] D33682
2021-12-09 08:34:04,020 [screen.py:show_panel()] - Unable to load panel main_menu
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 304, in show_panel
    self.panels[panel_name].initialize(panel_name, **kwargs)
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 38, in initialize
    leftpanel = self.create_left_panel()
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 192, in create_left_panel
    self.add_device(d)
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 107, in add_device
    temp.set_markup(self.format_temp(self._printer.get_dev_stat(device, "temperature")))
  File "/home/pi/KlipperScreen/ks_includes/screen_panel.py", line 53, in format_temp
    n = round(temp, places)
TypeError: type NoneType doesn't define __round__ method
2021-12-09 08:34:04,022 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel main_menu
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 304, in show_panel
    self.panels[panel_name].initialize(panel_name, **kwargs)
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 38, in initialize
    leftpanel = self.create_left_panel()
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 192, in create_left_panel
    self.add_device(d)
  File "/home/pi/KlipperScreen/panels/main_menu.py", line 107, in add_device
    temp.set_markup(self.format_temp(self._printer.get_dev_stat(device, "temperature")))
  File "/home/pi/KlipperScreen/ks_includes/screen_panel.py", line 53, in format_temp
    n = round(temp, places)
TypeError: type NoneType doesn't define __round__ method
2021-12-09 08:34:04,041 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe: {'objects': {'bed_mesh': ['profile_name', 'mesh_max', 'mesh_min', 'probed_matrix'], 'configfile': ['config'], 'display_status': ['progress', 'message'], 'fan': ['speed'], 'gcode_move': ['extrude_factor', 'gcode_position', 'homing_origin', 'speed_factor'], 'idle_timeout': ['state'], 'pause_resume': ['is_paused'], 'print_stats': ['print_duration', 'total_duration', 'filament_used', 'filename', 'state', 'message'], 'toolhead': ['homed_axes', 'estimated_print_time', 'print_time', 'position', 'extruder', 'max_accel', 'max_accel_to_decel', 'max_velocity', 'square_corner_velocity'], 'virtual_sdcard': ['file_position', 'is_active', 'progress'], 'webhooks': ['state', 'state_message'], 'extruder': ['target', 'temperature', 'pressure_advance', 'smooth_time'], 'heater_bed': ['target', 'temperature'], 'heater_generic chamber': ['target', 'temperature'], 'temperature_sensor Pi': ['target', 'temperature'], 'fan_generic filter_fan': ['speed'], 'heater_fan hotend_fan': ['speed'], 'heater_fan controller_fan': ['speed']}}
2021-12-09 08:34:04,045 [functions.py:logging_exception_handler()] - Uncaught exception <class 'TypeError'>: type NoneType doesn't define __round__ method
Traceback:   File "/home/pi/KlipperScreen/ks_includes/printer.py", line 344, in _update_temp_store
    self.tempstore[device][x].append(round(self.get_dev_stat(device, x[:-1]), 2))
NoneType: None
2021-12-09 08:34:04,973 [functions.py:logging_exception_handler()] - Uncaught exception <class 'TypeError'>: type NoneType doesn't define __round__ method
Traceback:   File "/home/pi/KlipperScreen/ks_includes/printer.py", line 344, in _update_temp_store
    self.tempstore[device][x].append(round(self.get_dev_stat(device, x[:-1]), 2))
NoneType: None```

##Versions:
Klipper: v0.10.0-173-gbea20278
Moonraker: v0.7.1-229-g54abd27
KlipperScreen: v0.2.0-0-g2a9e653

##Config:
My latest config can be found here https://github.com/th3fallen/voronConfig/

jordanruthe, 2021-12-09:
Can you please provide the full klipperscreen log file?

th3fallen, 2021-12-09:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7686157/KlipperScreen.log)
can do

LauOtero, 2021-12-10:
The same happens to me

th3fallen, 2021-12-10:
was hoping the update for klipperscreen today would fix it but no love :(

th3fallen, 2021-12-11:
So the patch today prevents the crash but now I have an unnamed temperature readout on the main menu with a value of 0?

alfrix, 2022-02-19:
@th3fallen a lot of changes have been made, is this fixed?

th3fallen, 2022-02-19:
Confirmed fixed

closed 2022-02-19 by th3fallen