KlipperScreen/KlipperScreen #365: Screen timeout not working

torbjornaxelsson, 2021-12-10:
My screen stays on indefinitely, regardless of timeout setting. I've attached my log file, see below. Timeout is currently set to 5 minutes.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7695681/KlipperScreen.log)

Hardware: Raspberry Pi 3B, 5" generic HDMI with USB touch
Software: FluiddPi with latest version of KlipperScreen

Grateful for any insight, thank you!

alfrix, 2021-12-11:
There is a problem with DPMS 
`[screen.py:check_dpms_state()] - DPMS State FAIL -> Showing KlipperScreen, Stopping DPMS Check´

in CLI try: `xset DPMS force off` and check if the screen turns off

torbjornaxelsson, 2021-12-11:
> There is a problem with DPMS `[screen.py:check_dpms_state()] - DPMS State FAIL -> Showing KlipperScreen, Stopping DPMS Check´
> 
> in CLI try: `xset DPMS force off` and check if the screen turns off

Thank you! That helped me troubleshoot. It turns out screen blanking was disabled on the raspberry pi, once I enabled it in raspi-config everything works. My display unfortunately displays a "No signal" message instead of turning off, but I believe that is a hardware issue. The same thing happens with the HDMI cable disconnected.

closed 2021-12-11 by torbjornaxelsson