KlipperScreen/KlipperScreen #382: Temperature graph on main screen

hapklaar, 2021-12-25:
The front page now shows a Main panel screenshot containing a very useful temperature graph, but I guess this is not available yet in the master branch right?

alfrix, 2021-12-25:
it is on master, just update, 

it's planned to also be available while printing and probably preheat panel

hapklaar, 2021-12-25:
I'm on the latest update using Kiauh (v0.1.6-175), but I my main screen doesn't show a graph.

alfrix, 2021-12-26:
again, that's not the latest update, i would use the integrated updater configuring moonraker, but if you insist on using kiauh, you need to figure out why it's not updating, but it is not an KS issue

hapklaar, 2021-12-26:
It seemed I had two installs of KlipperScreen on my system. Kiauh was updating one and I was running the other...

After changing KlipperScreen.service to point to the right location, I now get the latest and greatest including the temp graph.

I'm going to retest my other reported issues as they are probably all fixed as well.

Sorry about this, I feel stupid...

closed 2021-12-26 by hapklaar