KlipperScreen/KlipperScreen #403: [FR] A method to shutdown the host computer of KlipperScreen

kvikindi, 2022-01-05:
I've got KlipperScreen running off a secondary pi zero with a screen, and I'd love to be able to set a custom button to send a `sudo shutdown now` to be able to turn of the KS host after I've turned off the printer.

So far I've only managed to add a button, but I have no idea what the format is for sending a local command, as opposed to sending a shutdown command through moonraker to the printer.

Any suggestions would be welcomed!

jordanruthe, 2022-01-05:
There is a button in the system panel for shutting down the system.

kvikindi, 2022-01-05:
Thanks for the reply, but that button shuts down the connected system (in this case Klipper on a separate Pi Zero), not the host of the KlipperScreen.

What I'm trying to achieve is:
Turn off printer through actions - power - switch. (Sorted).
Once I've shut down the printer I arrive at the splash screen on KlipperScreen.
Have an option to send `sudo shutdown now` to the host of KS, not the remote.

Currently I only have the option of restarting/rebooting/shutting down the host with the Klipper install, but not the KlipperScreen pi itself.

Sorry if I sound convoluted - it's been a long day ;) - sidenote - absolutely love your software!

sloscotty, 2022-01-06:
I agree this would be a nice feature to have.  I am running KS on a pi that has no printer attached (and no moonraker installed).  It would be nice to be able to shut it down cleanly without having to ssh in to do a shutdown.

alfrix, 2022-05-23:
This is integrated on the splashscreen now, but only if moonraker is disconnected

PR #716 (2022-09-02): 0.2.6
fix #685
fix #403
fix #646
fix #671
fix #710
fix #714
fix #715

related to #699, the graph can be hidden, not equal but close to before
direct keypad access was implemented on the temp panel too #626
closed 2022-09-02 by alfrix