KlipperScreen/KlipperScreen #412: Network Screen Better, Though Not 100% Fixed

TehProgrammer, 2022-01-08:
I've noticed that, in the network screen, that there is always a nonsense connection.  Its SSID is a long string of repeating "\x00\x00\x00"....

I'm guessing this is my current connection?  As my SSID doesn't show up in the list.

randellhodges, 2022-01-08:
Are you broadcasting or hiding your SSID or have something called "Closed System Mode" on your router?

https://www.pclinuxos.com/forum/index.php?topic=115848.0

TehProgrammer, 2022-01-08:
> Are you broadcasting or hiding your SSID or have something called "Closed System Mode" on your router?
> 
> https://www.pclinuxos.com/forum/index.php?topic=115848.0

Broadcasting, and I'm not sure what closed system mode is.

However, we have WiFi 6 mesh system with a single SSID for both the 2.4 GHz and 5.8 GHz bands.

randellhodges, 2022-01-08:
No idea what "closed system mode" is either. I just saw it mentioned in that old forum post and thought I'd mention it.

alfrix, 2022-04-02:
should be fixed with 2a7acea86c3e1a4f6dc6e0bb85043e5f05c559df

closed 2022-04-02 by alfrix