KlipperScreen/KlipperScreen #427: [FR] Debounce Sliders

randellhodges, 2022-01-11:
This is more of an issue with the Gtk framework. It seems there is no slide/move end event so when you use the slider for something like the fans and slide from 0 to 100, it'll send like 100 gcodes over to klipper. It would be nice to only send the code when you are "done" sliding.

`self.devices[fan]['scale'].connect("value-changed", self.set_fan_speed, fan)`

I did some research and there might be something that can be done if you hook the change-value event instead. If that doesn't work, you can always use the classic javascript type debounce logic where you set/reset a timer on each value change and only trigger the callback when the timer expires.

I am willing to look into doing that and submit a PR if it is something that would be accepted

alfrix, 2022-01-24:
>  it'll send like 100 gcodes over to klipper.

i don't see this as an issue even in 8bit controllers, i know that mainsail and fluidd do apply the value only on release, but it's nice to be able to control the fan in real-time

i'm fine either way, i let Jordan decide

randellhodges, 2022-01-24:
I didn't mean to indicate there was a problem, that's why I prefixed it as a feature request. It could just be OCD, but the delay/ramping feeling I was getting while using it just annoyed me some. 

Probably not a big deal, but configuration values exposed as a slider would also get that many writes to the SD card as well.

closed 2022-02-17 by alfrix