KlipperScreen/KlipperScreen #429: No automatic refresh of temperature_fan

Tim2309, 2022-01-11:
I encountered a bug with the new temperature screen.
If you set a temperature_fan to 50C and change it through the Mainsail Webinterface, Klipperscreen didn't notice the change.
On extruder and bed heater the target temperature refresh works.

closed 2022-01-17 by jordanruthe

Tim2309, 2022-01-31:
The problem still exist if you use a custom gcode on the klipperscreen config to change the temperature

[preheat ABS]
# Temperature for the heated bed
bed: 110
# Temperature for the tools
extruder: 260
# Temperature for generic heaters
#heater_generic: 40
# Temperature controlled fans (temperature_fan in klipper config)
temperature_fan: 50
# optional GCode to run when the option is selected
gcode: SET_TEMPERATURE_FAN_TARGET temperature_fan=controller TARGET=30

alfrix, 2022-02-01:
The issue was that you are setting temperature fan to 50 then changing it in the gcode to 30
the target where applied but the ui was not updated.
7db75ec should fix this

PR #436 (2022-01-17): Temp_panel: put devices in a scroll
This way we can support unlimited devices, also avoids squishing the graph too much and makes the device button a bit larger so it's easier to press

master
[image]
pr
[image]

Also Fix #429
commit 5d3bae8f (2022-01-17): Temp_panel: put devices in a scroll (#436)

* Temp_panel: put devices in a scroll

* Tweak margins

* Tpanel: Fix whitespace in message

* Fix #429 temperature_fan not updating target