KlipperScreen/KlipperScreen #444: Fan Speed percentage is wrong on print screen

moelski, 2022-01-15:
Hi !

MainSail reports a fan speed at 33% where as klipperscreen reports only 13%. 
Another sample is Mainsail 100% and KS 40%. 

I assume there is a calculation error. Since the Fan speed is set in 255 steps. 
13% * 2.55 > 33%. 

regards 
    Dominik

randellhodges, 2022-01-15:
Which fan? With an older version of moonraker and older version of fluidd, I see 40% in both fluidd and klipperscreen.

fluidd v1.6.2
moonraker v0.7.1-63-g8676465

alfrix, 2022-01-16:
cannot reproduce configured hotend as fan_generic to adjust, also tested as heater_fan and it works ok too
[image]

moelski, 2022-01-17:
Try to reproduce it with my next prints.

moelski, 2022-01-17:
Hi !
Just make another print and could do some testing ... 
[image]
I set the fan speed in MainSail to ~40%. In Klipperscreen I see 16% (which is 40 / 2.55). 

[image]
The correst M106 command is send ... 

`Which fan?`
It´s the part fan ... 

Versions are the latest from all components: 
klipper
v0.10.0-234-g02d5f975
KlipperScreen
v0.2.0-34-g5d3bae8
moonraker
v0.7.1-300-g5f54a23
timelapse
v0.0.1-96-g90a277c
mainsail
v2.1.0-rc1

moelski, 2022-01-17:
Ok and just another print ... Just from the beginning ... 
[image]

MainSail schows 100% but the print screen only 40% in KS

alfrix, 2022-02-19:
can you post the fan config?

simonvez, 2022-03-13:
I have the same issue when I configure the  MAX_POWER  in klipper printer.cfg. For instance.. if I set max_power at 50%... when I wask 100%, it will show 50%... but mainsail will show 100%. Would be nice if klipper screen would not do that and just do like mainsail does and show the real percentage and not the calculated percentage based from what kipper max_power does.

alfrix, 2022-03-14:
i expected it to be some config, i can reproduce this behavior now, i should be able to fix it, thank you for your help

commit a893f476 (2022-03-15): Show fan speed according to max_power and off_below (#543)

fixes #444
closed 2022-03-15 by alfrix