KlipperScreen/KlipperScreen #446: Settings-Temperature - temps not updating

pgale, 2022-01-15:
During a print, if I go to settings-Temperature and say change the extruder temp, the temperature doesn't update in the list. Current temp or set temp.

Klipperscreen v0.2.0-30-gf796a72

alfrix, 2022-01-16:
cannot reproduce, temps update at different rates between the interfaces (not in sync), but they are the same.
[image]

pgale, 2022-01-16:
How do you get a screen shot from klipperscreen?

Many thanks for looking into this - I really like the screens but still definitely having an issue.

So I ran a long overnight job and I'm just starting another.

Fluidd shows the raspberry Pi temp of 49.2 but Klipperscreen shows 39.9 - this may be a different issue though I guess. It hasn't been 39.9 since I added the Pi temperature sensor into printer.cfg around 20 hours ago. But I've not restarted Klipperscreen (probably but can't remember).

In addition to that, I'm definitely seeing a problem with the extruder and bed temps in that screen alone. Now when I go in, the extruder is highlighted but nothing else - the temps do not update at all, neither the actual or set temps. If I press the increase or decrease (plus/minus) button, nothing updates but it does actually increase the set temp (if I go back to main screen, I see it increasing and the set point is correct). Also - even though just the extruder is highlighted, the plus/minus buttons are applied to both extruder and bed.

Kind of thinking that something has happened in the code that has thrown the whole screen into problems. I will restart klipperscreen after this job and reprot what then happens. If I see the issue, maybe i'll make a short video clip showing the problems?

Thanks - and thanks for the excellent klipperscreen, it really is fabulous :)

pgale, 2022-01-16:
So I just rebooted and the temperatures are now behaving themselves for now - I'll continue to test as there's definitely something there that causes the problem.

I have confirmed that the select function doesn't work as it should when first visiting the temperature screen - the extruder was the only selected item but the plus and minus buttons affected both extruder and bed. If I select and deselect the bed, it works as it should.

BTW, in the temp screen with the presets - ABS, PLA, FLEX, PETG - where are these defined? I wish to change these as they don't work for my temperatures.

alfrix, 2022-01-16:
Always get [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/) 

A screenshot would be a great plus, i use scrot from ssh or remmina trough a vnc connection.

https://klipperscreen.readthedocs.io/en/latest/Configuration/#preheat-options

alfrix, 2022-01-16:
and by the way, the selection and delta changes all work fine, it must been some memory corruption or who knows, no log = no info and makes this issue ticket not useful

pgale, 2022-01-16:
Yes sorry - I will get a log when I see the problem again. I've just installed VNC so will try and reproduce the problems and post that up here as soon as I can.

hapklaar, 2022-01-18:
I noticed as well that while in the temperature menu the temps are not always updating. I need to go back to main menu, then back into temps to get them updated. Can't reproduce this now either.

hapklaar, 2022-01-19:
fyi I need to restart klipperscreen service to get it working again.

@alfrix If you need any more info, let me know!

alfrix, 2022-01-20:
log and pictures, very useful, Thank you!
from the logs and the time of the captures, i believe the issue was originated when you saved the config after doing pid_calibrate, ks entered disconnected, and came back when the printer was ready, from my testing if ks enters disconnected or shutdown, it doesn't come back properly, (if you add devices to the printer.cfg they will not be shown, and other stuff), this is something that i have been trying to track-down and on my personal testing fork i've added a ks restart on those states, it's a work around that i believe Jordan will not want on the main repo

closed 2022-01-23 by alfrix

hapklaar, 2022-01-23:
Nice one!

commit 02c008de (2022-01-23): Temp_panel: Fix temps not updating after opening another instance
Fix #446