KlipperScreen/KlipperScreen #448: "Power" Menu?

Durahl, 2022-01-17:
Ever since switching over from a Mini 12864 Display to KlipperScreen I've been mourning the loss of being able to turn on my LEDs while at the printer but after having had a second look at the Documentation just now it appears for there to be a Power Panel controlling LED's:
[image]
Problem's just... I cannot find it?! 🤔

How my LEDs are defined in the `printer.cfg`:
```
[output_pin caselight]
# 24V Chamber Lighting - HE1 Connector (Optional)
pin: PA3
pwm: true
shutdown_value: 0
value: 1
cycle_time: 0.01
```

jordanruthe, 2022-01-17:
The power menu controls moonraker power devices. Please refer to https://moonraker.readthedocs.io/en/latest/configuration/#power

alfrix, 2022-01-18:
moonraker cannot be used with MCU pins

you should create a macro that turns on/off and add it to the menu
the only issue is that currently the options are not merged and you will have to copy the entire menu from defaults.conf

example note that more is a menu that might not exist in your config
```
[menu __main more LEDs]
name: LED
icon: light

[menu __main more LEDs LED_ON]
name: Turn LED On
icon: light
method: printer.gcode.script
params: {"script":"MACRO_LED_ON"}

[menu __main more LEDs LED_OFF]
name: Turn LED Off
icon: light
method: printer.gcode.script
params: {"script":"MACRO_LED_OFF"}
```
the menu system it's described in https://klipperscreen.readthedocs.io/en/latest/Configuration/#menu

closed 2022-02-19 by alfrix

mdriverch, 2022-02-23:
is it possible to put only the custom menus you want in the Klipperscreen.conf? or do i have to put the whole default.config? or can i load it with include?
i do not understand this in the documentation?