KlipperScreen/KlipperScreen #452: Failed to build numpy ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects

zrays, 2022-01-18:
Building wheels for collected packages: numpy
  Building wheel for numpy (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmp_5wh3jai
       cwd: /tmp/pip-install-ko_95y46/numpy_63ce16912c004cdbbadcc5f0f044c280
  Complete output (562 lines):
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/_philox.pyx
  Processing numpy/random/_sfc64.pyx
  Processing numpy/random/_common.pyx
  Processing numpy/random/mtrand.pyx
  Processing numpy/random/bit_generator.pyx
  Processing numpy/random/_mt19937.pyx
  Processing numpy/random/_bounded_integers.pyx.in
  Processing numpy/random/_generator.pyx
  Processing numpy/random/_pcg64.pyx
  Cythonizing sources
  non-existing path in 'numpy/distutils': 'site.cfg'
  running bdist_wheel
  running build
  running config_cc
  running config_fc
  running build_src
  creating build
  creating build/src.linux-armv7l-3.7
  creating build/src.linux-armv7l-3.7/numpy
  creating build/src.linux-armv7l-3.7/numpy/distutils
  Could not locate executable gfortran
  Could not locate executable f95
  Could not locate executable ifort
  Could not locate executable ifc
  Could not locate executable lf95
  Could not locate executable pgfortran
  Could not locate executable nvfortran
  Could not locate executable f90
  Could not locate executable f77
  Could not locate executable fort
  Could not locate executable efort
  Could not locate executable efc
  Could not locate executable g77
  Could not locate executable g95
  Could not locate executable pathf95
  Could not locate executable nagfor
  Could not locate executable frt
  don't know how to compile Fortran code on platform 'posix'
  creating build/src.linux-armv7l-3.7/numpy/core
  creating build/src.linux-armv7l-3.7/numpy/core/src
  creating build/src.linux-armv7l-3.7/numpy/core/src/npymath
  creating build/src.linux-armv7l-3.7/numpy/core/src/multiarray
  non-existing path in 'numpy/core': 'build/src.linux-armv7l-3.7/numpy/core/src/common'
  creating build/src.linux-armv7l-3.7/numpy/core/src/common
  creating build/src.linux-armv7l-3.7/numpy/core/src/npysort
  creating build/src.linux-armv7l-3.7/numpy/core/src/umath
  creating build/src.linux-armv7l-3.7/numpy/core/src/_simd
  creating build/src.linux-armv7l-3.7/numpy/linalg
  ### Warning:  Using unoptimized lapack ###
  ### Warning:  Using unoptimized lapack ###
  running build_py
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/numpy
  copying numpy/matlib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_pytesttester.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_globals.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/dual.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/conftest.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/ctypeslib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/setup.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/version.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_distributor_init.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/__init__.py -> build/lib.linux-armv7l-3.7/numpy
[... 479 lines cut ...]
      numpy/distutils/site.cfg file (section [blas_src]) or by setting
      the BLAS_SRC environment variable.
    if self._calc_info(blas):
  /tmp/pip-install-ko_95y46/numpy_63ce16912c004cdbbadcc5f0f044c280/numpy/distutils/system_info.py:1858: UserWarning:
      Lapack (http://www.netlib.org/lapack/) libraries not found.
      Directories to search for the libraries can be specified in the
      numpy/distutils/site.cfg file (section [lapack]) or by setting
      the LAPACK environment variable.
    return getattr(self, '_calc_info_{}'.format(name))()
  /tmp/pip-install-ko_95y46/numpy_63ce16912c004cdbbadcc5f0f044c280/numpy/distutils/system_info.py:1858: UserWarning:
      Lapack (http://www.netlib.org/lapack/) sources not found.
      Directories to search for the sources can be specified in the
      numpy/distutils/site.cfg file (section [lapack_src]) or by setting
      the LAPACK_SRC environment variable.
    return getattr(self, '_calc_info_{}'.format(name))()
  Warning: attempted relative import with no known parent package
  /tmp/pip-build-env-90_565f3/overlay/lib/python3.7/site-packages/setuptools/_distutils/dist.py:275: UserWarning: Unknown distribution option: 'define_macros'
    warnings.warn(msg)
  error: Command "arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithm_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_exponent_log.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.neon_vfpv4.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/abstractdtypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/alloc.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/arrayobject.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/arraytypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_coercion.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_method.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_assign_scalar.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_assign_array.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/arrayfunction_override.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/buffer.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/calculation.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/compiled_base.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/common.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/common_dtype.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/convert.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/convert_datatype.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/conversion_utils.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/ctors.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_strings.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_busday.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_busdaycal.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/descriptor.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dtypemeta.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dragon4.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dtype_transfer.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/einsum.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/einsum_sumprod.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/flagsobject.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/getset.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/hashdescr.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/item_selection.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/iterators.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/legacy_dtype_implementation.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/lowlevel_strided_loops.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/mapping.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/methods.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/multiarraymodule.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_templ.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_api.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_constr.o build/temp.linux-armv7l-3.7/ [... cut ...] mmon/npy_cpu_features.o -L/usr/lib -Lbuild/temp.linux-armv7l-3.7 -lnpymath -lm -o build/lib.linux-armv7l-3.7/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so" failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

e1w00db1ue5, 2022-01-18:
I also have this issue

`ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpl2dte9x7
       cwd: /tmp/pip-install-unxyzv67/numpy_dd5df13c49ca4e0395cffaacbb288e63
  Complete output (532 lines):
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/_philox.pyx
  Processing numpy/random/_sfc64.pyx
  Processing numpy/random/_common.pyx
  Processing numpy/random/mtrand.pyx
  Processing numpy/random/bit_generator.pyx
  Processing numpy/random/_mt19937.pyx
  Processing numpy/random/_bounded_integers.pyx.in
  Processing numpy/random/_generator.pyx
  Processing numpy/random/_pcg64.pyx
  Cythonizing sources
  Could not locate executable gfortran
  Could not locate executable f95
  Could not locate executable ifort
  Could not locate executable ifc
  Could not locate executable lf95
  Could not locate executable pgfortran
  Could not locate executable nvfortran
  Could not locate executable f90
  Could not locate executable f77
  Could not locate executable fort
  Could not locate executable efort
  Could not locate executable efc
  Could not locate executable g77
  Could not locate executable g95
  Could not locate executable pathf95
  Could not locate executable nagfor
  Could not locate executable frt
  don't know how to compile Fortran code on platform 'posix'
  non-existing path in 'numpy/distutils': 'site.cfg'
  running bdist_wheel
  running build
  running config_cc
  running config_fc
  running build_src
  creating build
  creating build/src.linux-armv7l-3.7
  creating build/src.linux-armv7l-3.7/numpy
  creating build/src.linux-armv7l-3.7/numpy/distutils
  creating build/src.linux-armv7l-3.7/numpy/core
  creating build/src.linux-armv7l-3.7/numpy/core/src
  creating build/src.linux-armv7l-3.7/numpy/core/src/npymath
  creating build/src.linux-armv7l-3.7/numpy/core/src/multiarray
  non-existing path in 'numpy/core': 'build/src.linux-armv7l-3.7/numpy/core/src/common'
  creating build/src.linux-armv7l-3.7/numpy/core/src/common
  creating build/src.linux-armv7l-3.7/numpy/core/src/npysort
  creating build/src.linux-armv7l-3.7/numpy/core/src/umath
  creating build/src.linux-armv7l-3.7/numpy/core/src/_simd
  creating build/src.linux-armv7l-3.7/numpy/linalg
  running build_py
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/numpy
  copying numpy/matlib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_pytesttester.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_globals.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/dual.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/conftest.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/ctypeslib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/setup.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/version.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_distributor_init.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/__init__.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_version.py -> build/lib.linux-armv7l-3.7/numpy
  copying build/src.linux-armv7l-3.7/numpy/__config__.py -> build/lib.linux-armv7l-3.7/numpy
[... 448 lines cut ...]
  creating build/temp.linux-armv7l-3.7/numpy/core/src/multiarray
  creating build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort
  creating build/temp.linux-armv7l-3.7/numpy/core/src/umath
  creating build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/common
  during RTL pass: expand
  numpy/core/src/umath/loops_trigonometric.dispatch.c.src: In function ‘FLOAT_sin_NEON_VFPV4’:
  numpy/core/src/umath/loops_trigonometric.dispatch.c.src:202:20: internal compiler error: in convert_move, at expr.c:218
   NPY_NO_EXPORT void NPY_CPU_DISPATCH_CURFX(FLOAT_@func@)
                      ^
  0xb6a80717 __libc_start_main
        /home/pi/dev/glibc-buster/csu/libc-start.c:308
  Please submit a full bug report,
  with preprocessed source if appropriate.
  Please include the complete backtrace with any bug report.
  See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
  lto-wrapper: fatal error: arm-linux-gnueabihf-gcc returned 1 exit status
  compilation terminated.
  /usr/bin/ld: error: lto-wrapper failed
  collect2: error: ld returned 1 exit status
  error: Command "arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithm_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_exponent_log.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.neon_vfpv4.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/abstractdtypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/alloc. [... cut ...] 3.7/numpy/core/src/common/python_xerbla.o -L/usr/lib/arm-linux-gnueabihf -L/usr/lib -Lbuild/temp.linux-armv7l-3.7 -lnpymath -lf77blas -lcblas -latlas -lf77blas -lcblas -lm -o build/lib.linux-armv7l-3.7/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so" failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
`

Stephane-Peel, 2022-01-18:
Hi, almost the same issue here : 

Building wheel for numpy (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 /home/pi/.KlipperScreen-env/lib/python3.7/site-packages                                                                                                                                  /pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmp6_zfxljx
       cwd: /tmp/pip-install-az5ur9mf/numpy_5a66bdcd77a345089074b2a681f68c21
  Complete output (532 lines):
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/_philox.pyx
  Processing numpy/random/_common.pyx
  Processing numpy/random/_generator.pyx
  Processing numpy/random/_sfc64.pyx
  Processing numpy/random/_pcg64.pyx
  Processing numpy/random/bit_generator.pyx
  Processing numpy/random/_mt19937.pyx
  Processing numpy/random/mtrand.pyx
  Processing numpy/random/_bounded_integers.pyx.in
  Cythonizing sources
  Could not locate executable gfortran
  Could not locate executable f95
  Could not locate executable ifort
  Could not locate executable ifc
  Could not locate executable lf95
  Could not locate executable pgfortran
  Could not locate executable nvfortran
  Could not locate executable f90
  Could not locate executable f77
  Could not locate executable fort
  Could not locate executable efort
  Could not locate executable efc
  Could not locate executable g77
  Could not locate executable g95
  Could not locate executable pathf95
  Could not locate executable nagfor
  Could not locate executable frt
  don't know how to compile Fortran code on platform 'posix'
  non-existing path in 'numpy/distutils': 'site.cfg'
  running bdist_wheel
  running build
  running config_cc
  running config_fc
  running build_src
  creating build
  creating build/src.linux-armv7l-3.7
  creating build/src.linux-armv7l-3.7/numpy
  creating build/src.linux-armv7l-3.7/numpy/distutils
  creating build/src.linux-armv7l-3.7/numpy/core
  creating build/src.linux-armv7l-3.7/numpy/core/src
  creating build/src.linux-armv7l-3.7/numpy/core/src/npymath
  creating build/src.linux-armv7l-3.7/numpy/core/src/multiarray
  non-existing path in 'numpy/core': 'build/src.linux-armv7l-3.7/numpy/core/src/common'
  creating build/src.linux-armv7l-3.7/numpy/core/src/common
  creating build/src.linux-armv7l-3.7/numpy/core/src/npysort
  creating build/src.linux-armv7l-3.7/numpy/core/src/umath
  creating build/src.linux-armv7l-3.7/numpy/core/src/_simd
  creating build/src.linux-armv7l-3.7/numpy/linalg
  running build_py
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/numpy
  copying numpy/version.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/__init__.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/conftest.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/matlib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/setup.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/ctypeslib.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_pytesttester.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_version.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_globals.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/dual.py -> build/lib.linux-armv7l-3.7/numpy
  copying numpy/_distributor_init.py -> build/lib.linux-armv7l-3.7/numpy
[... 450 lines cut ...]
  creating build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort
  creating build/temp.linux-armv7l-3.7/numpy/core/src/umath
  creating build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/common
  during RTL pass: expand
  numpy/core/src/umath/loops_trigonometric.dispatch.c.src: In function ‘FLOAT_sin_NEON_VFPV4’:
  numpy/core/src/umath/loops_trigonometric.dispatch.c.src:202:20: internal compiler error: in convert_move,                                                                                                                                   at expr.c:218
   NPY_NO_EXPORT void NPY_CPU_DISPATCH_CURFX(FLOAT_@func@)
                      ^
  0x76ae4717 __libc_start_main
        /home/pi/dev/glibc-buster/csu/libc-start.c:308
  Please submit a full bug report,
  with preprocessed source if appropriate.
  Please include the complete backtrace with any bug report.
  See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
  lto-wrapper: fatal error: arm-linux-gnueabihf-gcc returned 1 exit status
  compilation terminated.
  /usr/bin/ld: error: lto-wrapper failed
  collect2: error: ld returned 1 exit status
  error: Command "arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro bu                                                                                                                                  ild/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.neon.o bu                                                                                                                                  ild/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.neon.o                                                                                                                                   build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.o build                                                                                                             [... cut ...] y_umath.cpyt                                                                                                                                  hon-37m-arm-linux-gnueabihf.so" failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

toby5419, 2022-01-18:
Same issue please help

dhierholzer, 2022-01-18:
Same issue here. I have been working on it for a few hours now and still without any luck. Please help

alfrix, 2022-01-18:
it seems that numpy is broken?
add
`numpy==1.21.5`
to the top of the file `~/KlipperScreen/scripts/KlipperScreen-requirements.txt`
and reinstall

hybridindie, 2022-01-18:
you might mean 1.21.4, 1.21.5 is the current supported on MainsailOS and Python 3.7.x. 

I installed it using the following `KlipperScreen-requirement.txt` contents

```
 numpy==1.21.4
humanize==3.13.1
jinja2==3.0.3
matplotlib==3.5.1
netifaces==0.11.0
requests==2.26.0
websocket-client==1.2.3
pycairo==1.20.1
PyGObject==3.42.0
```

alfrix, 2022-01-18:
the lastest numpy release is 1.22.1, so i assumed the previous one, but ok

Stephane-Peel, 2022-01-18:
> you might mean 1.21.4, 1.21.5 is the current supported on MainsailOS and Python 3.7.x.
>
> I installed it using the following `KlipperScreen-requirement.txt` contents
>
> ```
>  numpy==1.21.4 [...]

Also worked for me !

Thx

dhierholzer, 2022-01-18:
Thank You!

Next question - Why Was it removed in teh first place?

cheers!

hybridindie, 2022-01-18:
> the lastest numpy release is 1.22.1, so i assumed the previous one, but ok

@alfrix 1.22.x drops support for python 3.7 which is what the MainsailOS has. No harm

@dhierholzer I don't think it's been removed, it's a requirement of matplotlib and pip will install the newest compatible version of a requirement, in this case it would be 1.21.5 which causes our problem.

toby5419, 2022-01-18:
when i add the line to the txt and reinstall, it change the txt to the old version anyone help?

e1w00db1ue5, 2022-01-18:
> the lastest numpy release is 1.22.1, so i assumed the previous one, but ok

this worked for me, 1.21.4 - thank you

closed 2022-01-19 by jordanruthe

mrwilby, 2022-01-19:
Pinning numpy to 1.21.4 prevents X11 from starting on my RPi. 

```
Jan 19 15:59:11 voron-black xinit[9952]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Jan 19 15:59:11 voron-black xinit[9952]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Jan 19 15:59:11 voron-black xinit[9953]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.63-v7l+
Jan 19 15:59:13 voron-black xinit[9970]: RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
Jan 19 15:59:13 voron-black xinit[9970]: Traceback (most recent call last):
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/KlipperScreen/screen.py", line 40, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     import matplotlib.pyplot
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/__init__.py", line 109, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from . import _api, _version, cbook, docstring, rcsetup
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/rcsetup.py", line 27, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from matplotlib.colors import Colormap, is_color_like
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/colors.py", line 56, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from matplotlib import _api, cbook, scale
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/scale.py", line 23, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from matplotlib.ticker import (
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/ticker.py", line 136, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from matplotlib import transforms as mtransforms
Jan 19 15:59:13 voron-black xinit[9970]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/transforms.py", line 46, in <module>
Jan 19 15:59:13 voron-black xinit[9970]:     from matplotlib._path import (
Jan 19 15:59:13 voron-black xinit[9970]: ImportError: numpy.core.multiarray failed to import
Jan 19 15:59:13 voron-black xinit[9951]: /usr/bin/xinit: connection to X server lost
Jan 19 15:59:13 voron-black xinit[9951]:
Jan 19 15:59:13 voron-black xinit[9951]: waiting for X server to shut down
Jan 19 15:59:13 voron-black xinit[9952]: (II) Server terminated successfully (0). Closing log file.
```

I had to upgrade numpy to 1.22.1 for X to start. Not sure what I am doing wrong, might be something specific to my environment.

alfrix, 2022-01-21:
No, you don't need 1.22.1 your issue is that you had 1.21.5 working (you are on bullseye not buster)
and matplotlib was compiled with the newer version, you need to uninstall matplotlib and reinstall after numpy 1.21.4
or just delete /home/pi/.KlipperScreen-env

if you don't do this downgrade, you will not be able to upgrade KS, the older numpy will have to stay for a while to give people time to upgrade their OSs

mrwilby, 2022-01-21:
Ok, good to know @alfrix , thanks for the info! I did delete .KlipperScreen-env multiple times, but no luck with (only) that.

alfrix, 2022-01-21:
i'm running bullseye too, the error is clearly matplotlib being compiled with a new version

try:
```
cd .KlipperScreen-env
source bin/activate  
pip cache purge
pip uninstall matplotlib
pip uninstall numpy
deactivate 
```
and then run the instaler

i have the following versions
```
(.KlipperScreen-env) ➜  .KlipperScreen-env python --version
Python 3.9.2
(.KlipperScreen-env) ➜  .KlipperScreen-env pip freeze
certifi==2021.10.8
charset-normalizer==2.0.9
cycler==0.11.0
fonttools==4.28.3
humanize==3.13.1
idna==3.3
Jinja2==3.0.3
kiwisolver==1.3.2
MarkupSafe==2.0.1
matplotlib==3.5.0
netifaces==0.11.0
numpy==1.21.4
packaging==21.3
Pillow==8.4.0
pkg_resources==0.0.0
pycairo==1.20.1
PyGObject==3.42.0
pyparsing==3.0.6
python-dateutil==2.8.2
requests==2.26.0
six==1.16.0
urllib3==1.26.7
websocket-client==1.2.1
```

PR #453 (2022-01-19): Add numpy to requirements to prevent unexpected issues
Fix #452