KlipperScreen/KlipperScreen #458: Unable to get klipperScreen to start

markcarroll, 2022-01-20:
Unable to get klipperScreen to start. WaveShare 5" Touchscreen. RPi 3B+ Ran through the troubleshooting steps. All are set as laid out in there. Still no luck. I just see the console screen on the display. 

```txt
$ cat /var/log/Xorg.0.log
[    11.535] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    11.535] Build Operating System: linux Raspbian
[    11.535] Current Operating System: Linux vera 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l
[    11.535] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=4301748d-02 rootfstype=ext4 fsck.repair=yes rootwait
[    11.535] Build Date: 04 January 2022  10:24:42AM
[    11.535] xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support) 
[    11.536] Current version of pixman: 0.40.0
[    11.536] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    11.536] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    11.536] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Jan 19 22:40:23 2022
[    11.563] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    11.578] (==) No Layout section.  Using the first Screen section.
[    11.578] (==) No screen section available. Using defaults.
[    11.578] (**) |-->Screen "Default Screen Section" (0)
[    11.578] (**) |   |-->Monitor "<default monitor>"
[    11.585] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    11.585] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    11.585] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    11.586] (**) Option "Debug" "dmabuf_capable"
[    11.586] (==) Automatically adding devices
[    11.586] (==) Automatically enabling devices
[    11.586] (==) Automatically adding GPU devices
[    11.594] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    11.637] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    11.637] 	Entry deleted from font path.
[    11.637] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    11.638] (==) ModulePath set to "/usr/lib/xorg/modules"
[    11.638] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    11.638] (II) Loader magic: 0x20cf40
[    11.638] (II) Module ABI versions:
[    11.638] 	X.Org ANSI C Emulation: 0.4
[    11.638] 	X.Org Video Driver: 24.1
[    11.638] 	X.Org XInput driver : 24.1
[    11.638] 	X.Org Server Extension : 10.0
[    12.046] (--) using VT number 2

[    12.046] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    12.051] (II) xfree86: Adding drm device (/dev/dri/card0)
[    12.052] (II) no primary bus or device found
[    12.052] 	falling back to /sys/devices/platform/soc/soc:gpu/drm/card0
[    12.052] (II) LoadModule: "glx"
[    12.065] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    12.229] (II) Module glx: vendor="X.Org Foundation"
[    12.229] 	compiled for 1.20.11, module version = 1.0.0
[    12.229] 	ABI class: X.Org Server Extension, version 10.0
[    12.229] (II) LoadModule: "fbturbo"
[... 232 lines cut ...]
[    15.722] (**) generic ft5x06 (79): Applying InputClass "libinput touchscreen catchall"
[    15.722] (II) Using input driver 'libinput' for 'generic ft5x06 (79)'
[    15.722] (**) generic ft5x06 (79): always reports core events
[    15.722] (**) Option "Device" "/dev/input/event1"
[    15.722] (**) Option "_source" "server/udev"
[    15.731] (II) event1  - generic ft5x06 (79): is tagged by udev as: Touchscreen
[    15.735] (II) event1  - generic ft5x06 (79): device is a touch device
[    15.736] (II) event1  - generic ft5x06 (79): device removed
[    15.850] (**) Option "config_info" "udev:/sys/devices/platform/soc/3f205000.i2c/i2c-11/i2c-10/10-0038/input/input8/event1"
[    15.850] (II) XINPUT: Adding extended input device "generic ft5x06 (79)" (type: TOUCHSCREEN, id 6)
[    15.850] (**) Option "AccelerationScheme" "none"
[    15.850] (**) generic ft5x06 (79): (accel) selected scheme none/0
[    15.850] (**) generic ft5x06 (79): (accel) acceleration factor: 2.000
[    15.850] (**) generic ft5x06 (79): (accel) acceleration threshold: 4
[    15.858] (II) event1  - generic ft5x06 (79): is tagged by udev as: Touchscreen
[    15.861] (II) event1  - generic ft5x06 (79): device is a touch device
[    21.671] (II) event0  - vc4: device removed
[    21.710] (II) event1  - generic ft5x06 (79): device removed
[    21.813] (II) UnloadModule: "libinput"
[    21.813] (II) UnloadModule: "libinput"
[    21.814] (II) UnloadModule: "libinput"
[    21.881] (II) Server terminated successfully (0). Closing log file.
```

Not sure what else to try.

alfrix, 2022-01-20:
If its an original display use this repo to install your screen: https://github.com/waveshare/LCD-show
If it's a clone use this: https://github.com/goodtft/LCD-show
reboot after installing the driver
check if switching fb0 to fb1 makes it work (or fb1 to fb0)
if you still cannot make it work try a different os image (raspbian lite - mainsailos - fluiddpi) and reinstall
as a last resort if you are still in trouble try the desktop version of raspbian and follow the steps in the docs to make it boot to console (https://klipperscreen.readthedocs.io/en/latest/Installation/#setup-raspberry-pi)

markcarroll, 2022-01-21:
There is no driver for the WaveShare 5 DSI but there was an [instruction](https://www.waveshare.com/wiki/5inch_DSI_LCD) to add a couple of lines to the `/boot/config.txt` which I did.  No luck. I then found a reference to needing to add the line `needs_root_rights=yes` into /etc/X11/Xwrapper.config which I did.  The screen then went blank a few times looking like it was trying to start. However it eventually goes back to the console screen.  Different errors this time:

```txt
Jan 20 16:47:24 vera xinit[3823]: (II) Server terminated successfully (0). Closing log file.
Jan 20 16:47:24 vera systemd[1]: KlipperScreen.service: Succeeded.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit KlipperScreen.service has successfully entered the 'dead' state.
Jan 20 16:47:24 vera systemd[1]: KlipperScreen.service: Consumed 3.115s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit KlipperScreen.service completed and consumed the indicated resources.
Jan 20 16:47:25 vera systemd[1]: KlipperScreen.service: Scheduled restart job, restart counter is at 118.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit KlipperScreen.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Jan 20 16:47:25 vera systemd[1]: Stopped KlipperScreen.
░░ Subject: A stop job for unit KlipperScreen.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit KlipperScreen.service has finished.
░░ 
░░ The job identifier is 8592 and the job result is done.
Jan 20 16:47:25 vera systemd[1]: KlipperScreen.service: Consumed 3.115s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit KlipperScreen.service completed and consumed the indicated resources.
Jan 20 16:47:25 vera systemd[1]: Started KlipperScreen.
░░ Subject: A start job for unit KlipperScreen.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit KlipperScreen.service has finished successfully.
░░ 
░░ The job identifier is 8592.
Jan 20 16:47:25 vera xinit[3851]: X.Org X Server 1.20.11
Jan 20 16:47:25 vera xinit[3851]: X Protocol Version 11, Revision 0
Jan 20 16:47:25 vera xinit[3851]: Build Operating System: linux Raspbian
Jan 20 16:47:25 vera xinit[3851]: Current Operating System: Linux vera 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 20>
Jan 20 16:47:25 vera xinit[3851]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=>
Jan 20 16:47:25 vera xinit[3851]: Build Date: 04 January 2022  10:24:42AM
Jan 20 16:47:25 vera xinit[3851]: xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support)
Jan 20 16:47:25 vera xinit[3851]: Current version of pixman: 0.40.0
Jan 20 16:47:25 vera xinit[3851]:         Before reporting problems, check http://wiki.x.org
Jan 20 16:47:25 vera xinit[3851]:         to make sure that you have the latest version.
Jan 20 16:47:25 vera xinit[3851]: Markers: (--) probed, (**) from config file, (==) default setting,
Jan 20 16:47:25 vera xinit[3851]:         (++) from command line, (!!) notice, (II) informational,
Jan 20 16:47:25 vera xinit[3851]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Jan 20 16:47:25 vera xinit[3851]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jan 20 16:47:25 2022
Jan 20 16:47:25 vera xinit[3851]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Jan 20 16:47:26 vera xinit[3852]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.63-v7+
Jan 20 16:47:28 vera xinit[3864]: RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
Jan 20 16:47:28 vera xinit[3864]: Traceback (most recent call last):
Jan 20 16:47:28 vera xinit[3864]:   File "/home/pi/KlipperScreen/screen.py", line 40, in <module>
Jan 20 16:47:28 vera xinit[3864]:     import matplotlib.pyplot
Jan 20 16:47:28 vera xinit[3864]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/__init__.p>
Jan 20 16:47:28 vera xinit[3864]:     from . import _api, _version, cbook, docstring, rcsetup
Jan 20 16:47:28 vera xinit[3864]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/rcsetup.py>
Jan 20 16:47:28 vera xinit[3864]:     from matplotlib.colors import Colormap, is_color_like
Jan 20 16:47:28 vera xinit[3864]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/colors.py">
[... 4 lines cut ...]
Jan 20 16:47:28 vera xinit[3864]:     from matplotlib import transforms as mtransforms
Jan 20 16:47:28 vera xinit[3864]:   File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/matplotlib/transforms>
Jan 20 16:47:28 vera xinit[3864]:     from matplotlib._path import (
Jan 20 16:47:28 vera xinit[3864]: ImportError: numpy.core.multiarray failed to import
Jan 20 16:47:29 vera xinit[3850]: /usr/bin/xinit: connection to X server lost
Jan 20 16:47:29 vera xinit[3850]: 
Jan 20 16:47:29 vera xinit[3850]: waiting for X server to shut down
Jan 20 16:47:29 vera xinit[3851]: (II) Server terminated successfully (0). Closing log file.
[... cut ...]
░░ The unit KlipperScreen.service has successfully entered the 'dead' state.
Jan 20 16:47:30 vera systemd[1]: KlipperScreen.service: Consumed 3.158s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit KlipperScreen.service completed and consumed the indicated resources.
```

I'm not a python expert but I tried `pip3 install -U numpy` but that didn't help.  It appears I have lumpy version 1.22.1.

Any other ideas?

alfrix, 2022-01-21:
with the log i can tell, that your issue is related with the recent issue with numpy 1.21.5, we had to downgrade because it doesn't work with debian buster, you are running bullseye, try this:

```
cd /home/pi/.KlipperScreen-env
source bin/activate  
pip cache purge
pip uninstall matplotlib
pip uninstall numpy
deactivate 
```
then run KlipperScreen-install

alfrix, 2022-01-21:
Fixed with d2419335b48cb43d18ef627faa5c1b8dabdcf792 just update or reinstall

closed 2022-01-21 by alfrix

markcarroll, 2022-01-21:
Fantastic - that did it. Thanks for the fix.