KlipperScreen/KlipperScreen #459: Stuck at connecting to printer

mparnapuu, 2022-01-20:
Hello!
I hope this is the right place, first time working with klipper and klipperscreen.
I installed klipper a few days ago, everything works through octoprint but when i tried to get my screen working with klipperscreen it just allways shows connecting to printer. i can restart klipper, do a firmware restart, and shutdown the system via the screen so it seems to semi work. i attatched the klipperscreen.log. 
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7906617/KlipperScreen.log)

alfrix, 2022-01-20:
error with your config...
you are missing a header probably [main]
i suggest you to delete `/home/pi/KlipperScreen.conf` and restart klipperscreen followed by an update because  you KS version is from 6 Oct 2021...

mparnapuu, 2022-01-20:
hmmm ok, 
tried deleting the config file, didnt help. wierd that the KS was old... maybe i installed it a month ago to test if pi functioned but not longer than that.
anyway. figured id just try reinstalling everything, maybe i messed up somewhere. Now i have a fresh octopi, klipper, moonraker, and klipperscreen.  the screen layout is a little bit different but the error is still the same, just shows connecting to printer
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7907980/KlipperScreen.log)

alfrix, 2022-01-20:
but now the issue is different
```
2022-01-20 19:24:50,764 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 891, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 109, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 121, in evaluate_state
    print_state = self.data['print_stats']['state'].lower()  # complete, paused, printing, standby
NoneType: None
```

mparnapuu, 2022-01-21:
issue seems to be the same, first error log: 
2022-01-20 15:47:13,284 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 822, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 104, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 114, in evaluate_state
    print_state = self.data['print_stats']['state'].lower()  # possible values: complete, paused, printing, standby
NoneType: None

2nd error log: 

2022-01-20 19:24:50,764 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 891, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 109, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 121, in evaluate_state
    print_state = self.data['print_stats']['state'].lower()  # complete, paused, printing, standby
NoneType: None

if these errors are the ones that caused this  then it seems to be the same issue? no? how can i get it working?

mparnapuu, 2022-01-21:
updated the printer.py that you changed. the error changed a little bit :

2022-01-21 16:40:10,094 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 891, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 109, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 121, in evaluate_state
    print_state = self.data['print_stats']['state']

alfrix, 2022-01-21:
pushed some changes, please try the new printer.py of the PR

mparnapuu, 2022-01-21:
new error :
2022-01-21 17:23:23,332 [functions.py:logging_exception_handler()] - Uncaught exception <class 'NameError'>: name 'data' is not defined
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 891, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 109, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 121, in evaluate_state
    if "print_stats" in data:
NoneType: None

changed  line 121 from if "print_stats" in data: 
to             if "print_stats" in self.data:

now error is same as before :

2022-01-21 17:25:42,218 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 891, in init_printer
    self.printer.process_update(data['result']['status'])

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 109, in process_update
    self.evaluate_state()

  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 122, in evaluate_state
    print_state = self.data['print_stats']['state']
NoneType: None

alfrix, 2022-01-21:
i know what is going on, i finally been able to reproduce the issue, you are missing
```
[virtual_sdcard]
path: ~/gcode_files
```
in printer.cfg

mparnapuu, 2022-01-21:
aahh god damit.. sorry!
yeah got it working... 
Thank you!

PR #461 (2022-01-23): Fix Startup without [virtual_sdcard]: evaluate_state
Fix #459
closed 2022-01-21 by mparnapuu