KlipperScreen/KlipperScreen #464: Unable to change preheat settings

AndersV209, 2022-01-22:
is there a way to change preheat settings without editing the default files (that get overwritten in an update) 

i have tried make the config file ~/klipper_config/KlipperScreen.conf
and added my own settings, but when after a restart the settings is the same as default

```
[preheat PLA]
bed = 60
extruder = 200

[preheat ABS]
bed = 70
extruder = 230

[preheat PETG]
bed = 70
extruder = 230

[preheat FLEX]
bed = 0
extruder = 210

```

i am running on RatOSwith latest update as of today 

[image]

alfrix, 2022-01-22:
it is highly possible that you have multiple KlipperScreen.conf files
search the home directory (usually: /home/pi/KlipperScreen.conf)
search the KlipperScreen directory (/home/pi/KlipperScreen/Klipperscreen.conf)
and lastly search if you have one with caps and one without (2 files KlipperScreen.conf and klipperscreen.conf)
delete all the instances and just leave 1 file
/tmp/KlipperScreen.log will show you what is reading
for example:
```
[config.py:get_config_file_location()] - Passed config file: /home/pi/KlipperScreen.conf
[config.py:get_config_file_location()] - Found configuration file at: /home/pi/klipper_config/klipperscreen.conf
[config.py:__init__()] - Config path location: /home/pi/klipper_config/klipperscreen.conf
```

AndersV209, 2022-01-22:
there was one in /root/KlipperScreen.conf

now i only have 1 

after I have deleted that it goes to the default config. and it didnt find my config file in the klipper config location 
~/klipper_config/KlipperScreen.conf

[image]

alfrix, 2022-01-23:
it will not work on /root/ that needs superuser access, why don't you place it in klipper_config?

AndersV209, 2022-01-23:
i know that is also the reason that i placed the config file i made in this location "~/klipper_config/KlipperScreen.conf" and not in root, the one in root was what it used before i deleted it.

alfrix, 2022-01-23:
there is one explanation to this, ran `sudo /KlipperScreen-install.sh`
that's a really bad idea...

install again as a regular user, read the docs: https://klipperscreen.readthedocs.io/en/latest/Installation/
or install with KIAUH

AndersV209, 2022-01-23:
i didnt install it, it came preinstalled on RatOS (os for running klipper on a ratrig v-core 3)

marcuswu, 2022-01-26:
I ran into this happening when I unintentionally added my temperature settings below the line:
`#~# --- Do not edit below this line. This section is auto generated --- #~`

Moving them above that line fixed the issue.

alfrix, 2022-02-19:
running KS as the root user, is not supported

closed 2022-02-19 by alfrix

alfrix, 2022-02-23:
It was an issue from Rat-OS https://github.com/Rat-OS/RatOS/commit/de6bfdfdf755958400b74ed0f1fe5e0874e6b17c

Follow their recommendation https://github.com/Rat-OS/RatOS/commit/35a1664dd0e3ce9982f624012ee3ccc3c52f81d1