KlipperScreen/KlipperScreen #475: [FR] Control view of resume buttons during macro loops

hawkeyexp, 2022-01-24:
Hello, i'm using some complex loops inside the start macro to jump during my initial steps like heating, calibration etc. which requires to pause the gcode during this steps and release it after the full pre print process is done.
During this loop where the system is paused KlipperScreen  shows the resume button which is by default correct but unwanted during this loops cause it can cause fails while loop is running in background and gcode is resumed. It would be great if we can hide the resume button by a command etc. which allows to hide the button starting with the pre print process and release it again if system is really printing :-)

alfrix, 2022-02-19:
I don't think this is valid, you should handle your case with the appropriate code in [gcode_macro PAUSE] and  [gcode_macro RESUME]

closed 2022-02-19 by alfrix

hawkeyexp, 2022-03-01:
It is valid and a common way to create complex print_start macros - have a look for example on code of alexz :-)

https://github.com/zellneralex/klipper_config