KlipperScreen/KlipperScreen #477: No Preheat Panel

TomMeyer90, 2022-01-25:
Hello,

In the new version is no longer available preheat panel

2022-01-25 08:02:12,227 [screen.py:_load_panel()] - Loading panel: preheat
2022-01-25 08:02:12,227 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/preheat.py
2022-01-25 08:02:12,228 [screen.py:_load_panel()] - Panel preheat does not exist
2022-01-25 08:02:12,228 [screen.py:show_panel()] - Unable to load panel prehea

[image]
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7931414/KlipperScreen.log)
t

randellhodges, 2022-01-25:
They preheat panel was removed:
https://github.com/jordanruthe/KlipperScreen/pull/393

alfrix, 2022-01-25:
> 
> The Preheat panel has been deprecated in favor of an all in one Temperature panel
> 
> Remove preheat panel from your Klipperscreen.conf or replace the occurrences of preheat with temperature

https://klipperscreen.readthedocs.io/en/latest/Changelog/

closed 2022-01-25 by alfrix