KlipperScreen/KlipperScreen #479: Is there a possible memory leak?

simpat1zq, 2022-01-25:
So a few weeks ago, my pi had been up for around a month. And I had a print that failed, and it just stopped right where it was. I looked at the system panel in Fluid and it was like 99% memory usage. I started restarted the services, and none of them affected it until I got to KlipperScreen. I restarted that one and it dropped down to a normal 34%. I just figured it was a one time thing and left it. 

However, I restarted the Pi a week ago, and already, it was using 65%. So I restarted KlipperScreen and it went back down. 

One thing I think it might be is that I switch between 5 printers on KlipperScreen, and I do it like 20 times a day. Not sure if that could be causing it. But just figured you might want to know about it.

simpat1zq, 2022-01-29:
So I think I know where the leak might be happening. It seems to happen in the files dialog. Everytime I open the files dialog after switching to a printer, it use up about 1% RAM, but switching to another printer doesn't release that RAM. So if I connect to a printer, tap print, then switch to another printer, tap print, and go back to the first, tap print, and then back to the other printer, tap print, it will increase the RAM usage by about 4%. And I was doing that about 5 times a day, which is why I saw all my RAM get used up and the system become unstable after about 2 weeks. 

And restarting KlipperScreen always fixes the issue.

alfrix, 2022-03-15:
tracemalloc:
```
/usr/lib/python3.7/json/decoder.py:353
/home/pi/KlipperScreen/ks_includes/files.py:72
/home/pi/KlipperScreen/ks_includes/files.py:87

```

simpat1zq, 2022-03-15:
Is there anything you want me to try regarding this?

commit 48b2ef51 (2022-03-20): Fix files and printer memory leaks fixes #479
closed 2022-03-20 by alfrix