KlipperScreen/KlipperScreen #481: Klipper screen stays on boot screen

Urminme, 2022-01-26:
Ive tried to edit the file 99-fbturbi.conf and it wont let me change it. I've tried uninstaling this then reinstalling it and nothing works there is always some kind of issue weither it be bad zip or kliper.screen masked or something. how is everyone having no problms with this i dont understand it

alfrix, 2022-01-26:
if it's masked do:
`sudo systemctl unmask KlipperScreen.service`
systemctl probably masked the service because something was wrong, it is strange

alfrix, 2022-01-26:
what hardware are you using, OS, log files and what did you do to get to this point, would be helpful

Urminme, 2022-01-26:
Sorry maybe I should close this now as I had to completely wipe my PI and start over, this time I used Kiauh as saw many people recommend this, which worked flawlessly, without need for any configurations or updates, just ask kiauh to download Klipperscreen, but I've noticed even though it works when I look at my updates klipper screen is not on there so I would have to SSH to update I would imagine. Not sure why I had days of problems with the install but it went from "bad zip file" errors to fully installing and not showing up to lastly saying It was masked, I gave up after almost a week of not being able to get help from Discord(#Klipper and #Klipperscreen), Voron discord or reddit so I just did a complete reinstall of everything. Thank you reaching out

alfrix, 2022-01-27:
i have ultimately made some changes to the installer to prevent mask issues, and created an uninstaller to easily start again.
using KIAUH it's ok it ultimately runs the installer too.
[The updater not showing KS is a missing configuration in moonraker.conf](https://klipperscreen.readthedocs.io/en/latest/Installation/#klipper-and-moonraker-installation)

```
[update_manager KlipperScreen]
type: git_repo
path: ~/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: ~/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh

```

Urminme, 2022-01-27:
Thank you for your response. Im just glad its working now, i may had done
something wrong at one point which lead to all the other issues, but a full
reinstall fixed all of it

closed 2022-01-27 by alfrix