KlipperScreen/KlipperScreen #483: Installation nicht möglich

Knoddel, 2022-01-30:
Ich bekomme die folgende Fehlermeldung beim Installieren von Klipperscreen.
Ob ich manuell oder über das kiauh Script installiere immer dasselbe. 
raspbian bullseye lite

`###### Installing KlipperScreen ...
Update package data
OK:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Paketlisten werden gelesen… Fertig
Checking for broken packages...
No broken packages
Installing KlipperScreen dependencies
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
fonts-freefont-ttf ist schon die neueste Version (20120503-10).
gcc ist schon die neueste Version (4:10.2.1-1+rpi1).
libatlas-base-dev ist schon die neueste Version (3.10.3-10+rpi1).
libgirepository1.0-dev ist schon die neueste Version (1.66.1-1+b1).
libopenjp2-7 ist schon die neueste Version (2.4.0-3).
matchbox-keyboard ist schon die neueste Version (0.2+git20160713-1).
pkg-config ist schon die neueste Version (0.29.2-1).
python3-dev ist schon die neueste Version (3.9.2-3).
python3-distutils ist schon die neueste Version (3.9.2-1).
python3-virtualenv ist schon die neueste Version (20.4.0+ds-2+deb11u1).
virtualenv ist schon die neueste Version (20.4.0+ds-2+deb11u1).
wireless-tools ist schon die neueste Version (30~pre9-13.1).
x11-xserver-utils ist schon die neueste Version (7.7+8).
xdotool ist schon die neueste Version (1:3.20160805.1-4).
xinit ist schon die neueste Version (1.4.0-1).
xinput ist schon die neueste Version (1.6.3-1).
gir1.2-gtk-3.0 ist schon die neueste Version (3.24.24-4+rpt3).
libcairo2-dev ist schon die neueste Version (1.16.0-5+rpt1).
xserver-xorg-video-fbturbo ist schon die neueste Version (1.20210602~095927+1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Creating virtual environment
created virtual environment CPython3.9.2.final.0-32 in 465ms
  creator CPython3Posix(dest=/home/pi/.KlipperScreen-env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==20.3.4, pkg_resources==0.0.0, setuptools==44.1.1, wheel==0.34.2
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./.KlipperScreen-env/lib/python3.9/site-packages (20.3.4)
Collecting pip
  Using cached https://www.piwheels.org/simple/pip/pip-22.0-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-22.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple

####################################################################################################ERROR: Exception:
Traceback (most recent call last):
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 165, in exc_logging_wrapper
    status = run_func(*args)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 339, in run
    requirement_set = resolver.resolve(
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 94, in resolve
    result = self._result = resolver.resolve(
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 481, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 348, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 172, in _add_to_criteria
    if not criterion.candidates:
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/resolvelib/structs.py", line 151, in __bool__
    return bool(self._sequence)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 155, in __bool__
    return any(self)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 143, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 44, in _iter_built
    for version, func in infos:
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 294, in iter_index_candidate_infos
    result = self._finder.find_best_candidate(
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/package_finder.py", line 868, in find_best_candidate
    candidates = self.find_all_candidates(project_name)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/package_finder.py", line 809, in find_all_candidates
    page_candidates = list(page_candidates_it)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/sources.py", line 134, in page_candidates
    yield from self._candidates_from_page(self._link)
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/package_finder.py", line 773, in process_project_url
    page_links = list(parse_links(html_page, self._use_deprecated_html5lib))
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 310, in wrapper_wrapper
    return list(fn(page, use_deprecated_html5lib))
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 350, in parse_links
    parser.feed(page.content.decode(encoding))
  File "/usr/lib/python3.9/html/parser.py", line 110, in feed
    self.goahead(0)
  File "/usr/lib/python3.9/html/parser.py", line 178, in goahead
    k = self.parse_html_declaration(i)
  File "/usr/lib/python3.9/html/parser.py", line 269, in parse_html_declaration
    self.handle_decl(rawdata[i+2:gtpos])
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 405, in handle_decl
    self._raise_error()
  File "/home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 426, in _raise_error
    raise ValueError(
ValueError: HTML doctype missing or incorrect. Expected <!DOCTYPE html>.

If you believe this error to be incorrect, try passing the command line option --use-deprecated=html5lib and please leave a comment on the pip issue at https://github.com/pypa/pip/issues/10825.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.`

alfrix, 2022-01-30:
Oh no they broke everything, what were they thinking?

alfrix, 2022-01-30:
I addressed this with 588869aa57909e22b8741884308af514add67fcf but this is an upstream issue, they will most likely fix it shortly

For now:
delete /home/pi/.KlipperScreen-env (if you know what you are doing you can just downgrade pip in the env)
and reinstall

alfrix, 2022-01-30:
f2d642b is the workaround

Knoddel, 2022-01-31:
@alfrix
Thanks for your support
With the new kiauh script everything works fine

closed 2022-01-31 by Knoddel

guiguijke, 2022-01-31:
Sorry to reopen this, but my install isn't working too ;

```
pi@mainsailos:~/KlipperScreen $ ./scripts/KlipperScreen-install.sh
Update package data
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Checking for broken packages...
No broken packages
Installing KlipperScreen dependencies
Reading package lists... Done
Building dependency tree
Reading state information... Done
fonts-freefont-ttf is already the newest version (20120503-9).
gcc is already the newest version (4:8.3.0-1+rpi2).
libatlas-base-dev is already the newest version (3.10.3-8+rpi1).
libgirepository1.0-dev is already the newest version (1.58.3-2).
libopenjp2-7 is already the newest version (2.3.0-2+deb10u2).
matchbox-keyboard is already the newest version (0.1+svn20080916-12).
pkg-config is already the newest version (0.29-6).
python3-dev is already the newest version (3.7.3-1).
python3-distutils is already the newest version (3.7.3-1).
python3-virtualenv is already the newest version (15.1.0+ds-2).
virtualenv is already the newest version (15.1.0+ds-2).
wireless-tools is already the newest version (30~pre9-13).
xdotool is already the newest version (1:3.20160805.1-4).
xinit is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.2-1).
gir1.2-gtk-3.0 is already the newest version (3.24.5-1+rpt2).
libcairo2-dev is already the newest version (1.16.0-4+rpt1).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xserver-xorg-video-fbturbo is already the newest version (1.20190107~114053).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Creating virtual environment
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Exception:
Traceback (most recent call last):
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 165, in exc_logging_wrapper
    status = run_func(*args)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 340, in run
    reqs, check_supported_wheels=not options.target_dir
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 95, in resolve
    collected.requirements, max_rounds=try_to_avoid_resolution_too_deep
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 481, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 348, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 172, in _add_to_criteria
    if not criterion.candidates:
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_vendor/resolvelib/structs.py", line 151, in __bool__
    return bool(self._sequence)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 155, in __bool__
    return any(self)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 143, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 44, in _iter_built
    for version, func in infos:
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 297, in iter_index_candidate_infos
    hashes=hashes,
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 868, in find_best_candidate
    candidates = self.find_all_candidates(project_name)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 809, in find_all_candidates
    page_candidates = list(page_candidates_it)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/sources.py", line 134, in page_candidates
    yield from self._candidates_from_page(self._link)
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 773, in process_project_url
    page_links = list(parse_links(html_page, self._use_deprecated_html5lib))
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/collector.py", line 310, in wrapper_wrapper
    return list(fn(page, use_deprecated_html5lib))
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/collector.py", line 350, in parse_links
    parser.feed(page.content.decode(encoding))
  File "/usr/lib/python3.7/html/parser.py", line 111, in feed
    self.goahead(0)
  File "/usr/lib/python3.7/html/parser.py", line 179, in goahead
    k = self.parse_html_declaration(i)
  File "/usr/lib/python3.7/html/parser.py", line 270, in parse_html_declaration
    self.handle_decl(rawdata[i+2:gtpos])
[... cut ...]
  File "/home/pi/.KlipperScreen-env/lib/python3.7/site-packages/pip/_internal/index/collector.py", line 427, in _raise_error
    "HTML doctype missing or incorrect. Expected <!DOCTYPE html>.\n\n"
ValueError: HTML doctype missing or incorrect. Expected <!DOCTYPE html>.

If you believe this error to be incorrect, try passing the command line option --use-deprecated=html5lib and please leave a comment on the pip issue at https://github.com/pypa/pip/issues/10825.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.
```

alfrix, 2022-02-01:
you probably updated during the weekend, update the klipperscreen repo and reinstall, if it fails again run the Uninstaller and then reinstall

guiguijke, 2022-02-01:
> you probably updated during the weekend, update the klipperscreen repo and reinstall, if it fails again run the Uninstaller and then reinstall

You're awesome , thanks =)