KlipperScreen/KlipperScreen #491: BiquBX 7" Display (BTT 7" TouchDisplay)

ZeroCoolAT, 2022-02-05:
Hey Jordan,

I'm about to use the biquBX with Klipper and want to use your awesome GUI instead of octodash.
Unfortunately it isn't working properly right now.

I setup the following values in my `/boot/config.txt`

```
max_usb_current=1
config_hdmi_boost=7
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

[image]

alfrix, 2022-02-06:
it's probably a configuration issue, i would start by grabbing another sdcard and install octobtt, when i see that working correctly copy the config.txt to the sdcard where you installed klipperscreen, it may need more than that, but you have to start somewhere

nicedevil007, 2022-02-07:
Maybe we can reflash the firmware of the TFT?

We will try this afternoon

alfrix, 2022-03-11:
Closing because of inactivity

closed 2022-03-11 by alfrix