KlipperScreen/KlipperScreen #492: [FR] Add ability to add Power button on 'shutdown' splash screen

hapklaar, 2022-02-06:
Can we have direct access to the power button on the splash screen which is shown when Klipper was shutdown?

This screen now shows 'Restart', 'Firmware restart' and 'Menu'. To power on the printer, I need to select Menu > Power > slide the power button. It would be nice and efficient to have the power button right on the splash screen, so I can power on the printer in a single act.

alfrix, 2022-02-17:
the splash screen should show power the power device should be named starting with printer
https://github.com/jordanruthe/KlipperScreen/blob/062593506292e64b987bf075e143fac60b4603e3/panels/splash_screen.py#L81

hapklaar, 2022-02-17:
Oh sweet. Renamed my power device, works like a charm!

closed 2022-02-17 by hapklaar