KlipperScreen/KlipperScreen #517: Screen button Mapping issue

behold81, 2022-02-21:
I’m using a small screen so accuracy is key to be usable. My touch is 100% calibrated and testing shows with a stylus it’s very accurate.  When I test klipper screen buttons every button is spot on to the very edge then non responsive once off the button as expected until you use the left set of buttons. Up to about 1/3 in from the right activates the button to the right and not the back,home etc button that I’m trying to press.  Is there a mapping fault? 

It makes it hard to go back a screen without hitting the main buttons instead. I have to really cram my finger to the edge to avoid any miss hit.

i tested this accuracy with a stylus. 

It should be like the yellow not the blue on the pic.  
[image]

alfrix, 2022-02-21:
it's definitively not like the blue area, i recently gave it a little bit more of margin (https://github.com/jordanruthe/KlipperScreen/commit/45d195bcd237c6e6f89274334b152acd2da743c8) so i you need to check your version first

behold81, 2022-02-22:
I’ll check tomorrow and see. Thanks.

behold81, 2022-02-22:
> it's definitively not like the blue area, i recently gave it a little bit more of margin ([45d195b](https://github.com/jordanruthe/KlipperScreen/commit/45d195bcd237c6e6f89274334b152acd2da743c8)) so i you need to check your version first

So I upgraded to V95 and its maybe a fraction better but its still not right on that edge. I am using a stylus to test accuracy. It might be screen setup but my calibration test edge to edge tracks correct. 

Screen is a 480x320 MKS T35 v2

alfrix, 2022-02-23:
i also have a 3.5" 480x320 and not having any issues after the fix, 
if you keep pressed the button in some themes you will see it highlighted
[image]

FYI, the calibration output from xinput-calibrator should be saved to /etc/X11/xorg.conf.d/99-calibration.conf and KS should be restarted

behold81, 2022-02-24:
So its still a problem. not sure why. maybe a setting on my touch screen. If I use TS_Calibrate/Test then it tracks my stylus to every corner with out issue. 
I run the xinput-calibrator and put in the results. (have actually tried doing this 10 times and then taking the average positions and used these. 

If I click a button and hold and then move my stylus watching the button depress/unpress all buttons seem to stop/start on the border of the button as expected both up/down and Left/right. 
However the most left 2 are always overlapping the side bar. its roughly 1/3 over. and if I select the side button it depresses about 2/3. Not sure what the issue is.

alfrix, 2022-03-16:
i have confirmed with a waveshare35b-v2 screen that the issue is the hardware calibration, xinput-calibrator output doesn't work, i followed https://wiki.archlinux.org/title/Talk:Calibrating_Touchscreen#Libinput%5Fbreaks%5Fxinput%5Fcalibrator

i found a script that will calculate the values and made some mods to it
```
#!/usr/bin/env python3
# -*- coding: UTF-8 -*-
import sys

def usage():
    info = "Usage: " + sys.argv[0]
    info += " <screen_width> <screen_height>"
    info += " <click_0_x> <click_0_y>"
    info += " <click_3_x> <click_3_y>"
    print(info)
    print("\tScreen width/hight by the command 'xrandr|grep screen'")
    print("\tClick x/y by the command 'xinput_calibrator -v'")
    sys.exit(0)

def convert(screen_x, screen_y, c0_x, c0_y, c3_x, c3_y):
    a = (screen_x * 6 / 8) / (c3_x - c0_x)
    c = ((screen_x / 8) - (a * c0_x)) / screen_x
    e = (screen_y * 6 / 8) / (c3_y - c0_y)
    f = ((screen_y / 8) - (e * c0_y)) / screen_y

    print("Option \"CalibrationMatrix\" \"%.2f 0 %.2f 0 %.2f %.2f 0 0 1\"" % (a,c,e, f))

if __name__ == "__main__":
    if len(sys.argv) != 7:
        usage()

    convert(int(sys.argv[1]), int(sys.argv[2]), int(sys.argv[3]),
    int(sys.argv[4]),int(sys.argv[5]),int(sys.argv[6]))
```
and that should  be in the corresponding section of 99-calibration.conf and will take effect after a ks restart

but it's still terribly inaccurate, you can check it by entering the console panel if you touch the screen there will be a cursor

one of the downsides of this cheap screen

closed 2022-03-16 by alfrix

alfrix, 2024-03-15:
i found a reliable way to calibrate the screen using a tool called xtcal i detailed the process in:

https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Touch_issues/#touch-calibration