KlipperScreen/KlipperScreen #524: no signal on screen

mxyzptlk-cn, 2022-02-27:
I'm running MainsailOS 0.6.1 with rpi4b with a 7 inch touch display, but screen shows no image, only says no signal.
Here are some related log and info, pls help. Tks.

pi@mainsail:~ $ systemctl status KlipperScreen
● KlipperScreen.service - KlipperScreen
   Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2022-02-27 23:59:23 CST; 4s ago
 Main PID: 2412 (xinit)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/KlipperScreen.service
           └─2412 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py

Feb 27 23:59:23 mainsail xinit[2412]: (EE)
Feb 27 23:59:23 mainsail xinit[2412]: Fatal server error:
Feb 27 23:59:23 mainsail xinit[2412]: (EE) no screens found(EE)
Feb 27 23:59:23 mainsail xinit[2412]: (EE)
Feb 27 23:59:23 mainsail xinit[2412]: Please consult the The X.Org Foundation support
Feb 27 23:59:23 mainsail xinit[2412]:          at http://wiki.x.org
Feb 27 23:59:23 mainsail xinit[2412]:  for help.
Feb 27 23:59:23 mainsail xinit[2412]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Feb 27 23:59:23 mainsail xinit[2412]: (EE)
Feb 27 23:59:23 mainsail xinit[2412]: (EE) Server terminated with error (1). Closing log file.

Feb 27 23:59:54 mainsail systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit KlipperScreen.service has entered the 'failed' state with result 'exit-code'.
Feb 27 23:59:55 mainsail systemd[1]: KlipperScreen.service: Service RestartSec=1s expired, scheduling restart.
Feb 27 23:59:55 mainsail systemd[1]: KlipperScreen.service: Scheduled restart job, restart counter is at 30.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit KlipperScreen.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Feb 27 23:59:55 mainsail systemd[1]: Stopped KlipperScreen.
-- Subject: A stop job for unit KlipperScreen.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit KlipperScreen.service has finished.
--
-- The job identifier is 2558 and the job result is done.
Feb 27 23:59:55 mainsail systemd[1]: Started KlipperScreen.
-- Subject: A start job for unit KlipperScreen.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit KlipperScreen.service has finished successfully.
--
-- The job identifier is 2558.
Feb 27 23:59:55 mainsail xinit[2614]: X.Org X Server 1.20.4
Feb 27 23:59:55 mainsail xinit[2614]: X Protocol Version 11, Revision 0
Feb 27 23:59:55 mainsail xinit[2614]: Build Operating System: Linux 5.4.0-72-generic armv8l Raspbian
Feb 27 23:59:55 mainsail xinit[2614]: Current Operating System: Linux mainsail 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l
Feb 27 23:59:55 mainsail xinit[2614]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:87:BE:3F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=bd9dbfd3-02 rootfstype=ext4 fsck.repair=yes rootwait
Feb 27 23:59:55 mainsail xinit[2614]: Build Date: 04 January 2022  10:05:34AM
Feb 27 23:59:55 mainsail xinit[2614]: xorg-server 2:1.20.4-1+rpt4+deb10u4 (https://www.debian.org/support)
Feb 27 23:59:55 mainsail xinit[2614]: Current version of pixman: 0.36.0
Feb 27 23:59:55 mainsail xinit[2614]:         Before reporting problems, check http://wiki.x.org
Feb 27 23:59:55 mainsail xinit[2614]:         to make sure that you have the latest version.
Feb 27 23:59:55 mainsail xinit[2614]: Markers: (--) probed, (**) from config file, (==) default setting,
Feb 27 23:59:55 mainsail xinit[2614]:         (++) from command line, (!!) notice, (II) informational,
Feb 27 23:59:55 mainsail xinit[2614]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Feb 27 23:59:55 mainsail xinit[2614]: (==) Log file: "/var/log/Xorg.0.log", Time: Sun Feb 27 23:59:55 2022
Feb 27 23:59:55 mainsail xinit[2614]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 27 23:59:55 mainsail xinit[2614]: (EE)
Feb 27 23:59:55 mainsail xinit[2614]: Fatal server error:
Feb 27 23:59:55 mainsail xinit[2614]: (EE) no screens found(EE)
Feb 27 23:59:55 mainsail xinit[2614]: (EE)
Feb 27 23:59:55 mainsail xinit[2614]: Please consult the The X.Org Foundation support
Feb 27 23:59:55 mainsail xinit[2614]:          at http://wiki.x.org
Feb 27 23:59:55 mainsail xinit[2614]:  for help.
Feb 27 23:59:55 mainsail xinit[2614]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Feb 27 23:59:55 mainsail xinit[2614]: (EE)
Feb 27 23:59:55 mainsail xinit[2614]: (EE) Server terminated with error (1). Closing log file.

pi@mainsail:~ $ ls /dev/tty[0-9]
/dev/tty0  /dev/tty1  /dev/tty2  /dev/tty3  /dev/tty4  /dev/tty5  /dev/tty6  /dev/tty7  /dev/tty8  /dev/tty9

pi@mainsail:~ $ cat /etc/group | grep tty
tty:x:5:pi

pi@mainsail:~ $ cat /etc/X11/Xwrapper.config
allowed_users=anybody

mxyzptlk-cn, 2022-03-02:
can't get it to work, so I tried using fluiddos, now it works.

alfrix, 2022-03-02:
no screen found means, you didn't install the proper drivers, or the screen cable or physical interface wasn't installed properly
don't blame mainsailos

closed 2022-03-02 by alfrix