KlipperScreen/KlipperScreen #529: Black screen while printing

dadonali, 2022-03-02:
I have the problem that the display goes black for no apparent reason during printing and then can no longer be operated. Only a restart of the KlipperScreen service solves the problem. The display can then be operated again

here is the klipperscreen log:

022-03-01 20:29:08,234 [config.py:__init__()] - Detected language: None
2022-03-01 20:29:08,244 [screen.py:__init__()] - Found wireless interfaces: ['wlan0']
2022-03-01 20:29:08,246 [wifi.py:run()] - Setting up wifi event loop
2022-03-01 20:29:08,248 [screen.py:__init__()] - OS Language: en_GB.UTF-8
2022-03-01 20:29:08,554 [screen.py:__init__()] - Font settings: Sans 10
2022-03-01 20:29:08,603 [screen.py:__init__()] - Screen resolution: 800x480
2022-03-01 20:29:08,604 [KlippyGtk.py:__init__()] - img width: 50 height: 51
2022-03-01 20:29:08,637 [base_panel.py:__init__()] - Button scale: [1.2, 1.4]
2022-03-01 20:29:09,012 [screen.py:_load_panel()] - Loading panel: splash_screen
2022-03-01 20:29:09,012 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2022-03-01 20:29:09,039 [screen.py:show_panel()] - Attaching panel splash_screen
2022-03-01 20:29:09,043 [screen.py:show_panel()] - Showing back. count: 0
2022-03-01 20:29:09,051 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-03-01 20:29:09,077 [splash_screen.py:show_restart_buttons()] - Power devices: []
2022-03-01 20:29:09,109 [screen.py:set_screenblanking_timeout()] - Changing power save to: off
2022-03-01 20:29:09,255 [screen.py:__init__()] - Printers: [{'Printer': {'moonraker_host': '127.0.0.1', 'moonraker_port': '7125', 'moonraker_api_key': ''}}]
2022-03-01 20:29:09,256 [screen.py:connect_printer()] - Connecting to printer: Printer
2022-03-01 20:29:09,275 [screen.py:show_panel()] - Attaching panel splash_screen
2022-03-01 20:29:09,280 [screen.py:show_panel()] - Showing back. count: 0
2022-03-01 20:29:09,286 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-03-01 20:29:09,310 [splash_screen.py:show_restart_buttons()] - Power devices: []
2022-03-01 20:29:09,316 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-03-01 20:29:09,327 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,353 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2022-03-01 20:29:09,357 [screen.py:connect_printer()] - Found power devices: False
2022-03-01 20:29:09,359 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-03-01 20:29:09,363 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,373 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /server/info HTTP/1.1" 200 640
2022-03-01 20:29:09,376 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-03-01 20:29:09,381 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,393 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 200 375
2022-03-01 20:29:09,396 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-03-01 20:29:09,401 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,411 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 200 46
2022-03-01 20:29:09,415 [screen.py:connect_printer()] - Connected to printer: Printer
2022-03-01 20:29:09,427 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-03-01 20:29:09,427 [KlippyWebsocket.py:on_open()] - Self.connected = False
2022-03-01 20:29:09,457 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-03-01 20:29:09,463 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,473 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 200 375
2022-03-01 20:29:09,477 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks
2022-03-01 20:29:09,484 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,685 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks HTTP/1.1" 200 30630
2022-03-01 20:29:09,694 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-03-01 20:29:09,702 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,737 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2022-03-01 20:29:09,745 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2022-03-01 20:29:09,756 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2022-03-01 20:29:09,925 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/objects/query?configfile HTTP/1.1" 200 24870
2022-03-01 20:29:09,938 [printer.py:reinit()] - Moonraker object status: {'configfile': {'warnings': [], 'config': {'gcode_macro BLT_pos': {'gcode': '\nG90\nG1 X170 Y95 F2000\nG91', 'description': 'BLTouch Touchmode'}, 'pause_resume': {}, 'gcode_macro PID_BED': {'gcode': '\nPID_CALIBRATE HEATER=heater_bed TARGET=60'}, 'safe_z_home': {'home_xy_position': '175.5,100', 'z_hop': '10'}, 'gcode_macro M600': {'gcode': '\n{% set X = params.X|default(50)|float %}\n{% set Y = params.Y|default(0)|float %}\n{% set Z = params.Z|default(10)|float %}\nSAVE_GCODE_STATE NAME=M600_state\nPAUSE\nG91\nG1 E-.8 F2700\nG1 Z{Z}\nG90\nG1 X{X} Y{Y} F3000\nG91\nG1 E-50 F1000\nRESTORE_GCODE_STATE NAME=M600_state'}, 'printer': {'max_accel': '1500', 'max_z_accel': '100', 'max_velocity': '150', 'max_z_velocity': '5', 'kinematics': 'cartesian', 'max_accel_to_decel': '1500'}, 'gcode_macro END_PRINT': {'gcode': '\nTURN_OFF_HEATERS                         ; Turn off bed and nozzle\nG91                                      ; Relative positioning\nG1 E-1 F3000                             ; Retract\nG1 X-0.5 Y-0.5 Z5 E-5                    ; Move a bit and retract filament even more\nM107                                     ; Turn off part fan\nG91                                      ; Absolute positioning\nG1 E-1 F300                              ;retract the filament a bit before lifting the nozzle, to release some of the pressure\nG1 Z+0.5 E-5                             ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more\nG28 X0                                   ;Y100 ;move X/Y to min endstops, so the head is out of the way\nM84                                      ; Steppers off\nG90                                      ; Absolute positioning\nM117 Print done                          ; Send finish to display', 'description': 'All what needs to be done at print end'}, 'gcode_macro PA_Calibration': {'gcode': '\nSET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500\nTUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020', 'description': 'Adjust Velocity and PA parameters for bowden extruder'}, 'z_tilt': {'retries': '7', 'horizontal_move_z': '7', 'z_positions': '\n-30,135\n275,135', 'retry_tolerance': '0.015', 'points': '\n60,105\n260,105', 'speed': '140'}, 'heater_bed': {'control': 'pid', 'pid_kp': '69.678', 'sensor_pin': 'PB0', 'heater_pin': 'PA4', 'sensor_type': 'EPCOS 100K B57560G104F', 'pid_kd': '909.293', 'pid_ki': '1.335', 'min_temp': '0', 'max_temp': '110'}, 'temperature_sensor Raspi': {'min_temp': '0', 'sensor_type': 'temperature_host', 'max_temp': '100'}, 'bed_mesh default': {'tension': '0.2', 'min_x': '13.0', 'min_y': '7.0', 'y_count': '6', 'mesh_y_pps': '3', 'x_count': '6', 'version': '1', 'algo': 'bicubic', 'mesh_x_pps': '2', 'max_y': '227.0', 'max_x': '206.95', 'points': '\n-0.095000, -0.025000, -0.007500, 0.062500, 0.045000, -0.037500\n-0.050000, -0.027500, 0.010000, 0.027500, -0.010000, 0.030 [... cut ...] 0.1, 'dir_pin': '!PB3', 'pid_kp': 28.128, 'instantaneous_corner_velocity': 1.0, 'full_steps_per_rotation': 200, 'heater_pin': 'PA1', 'smooth_time': 1.0, 'pid_kd': 140.29, 'pid_ki': 1.41, 'filament_diameter': 1.75, 'sensor_pin': 'PC4', 'nozzle_diameter': 0.4, 'pressure_advance_smooth_time': 0.04, 'step_pin': 'PB4', 'inline_resistor': 0.0, 'max_power': 1.0, 'microsteps': 16, 'max_temp': 260.0, 'max_extrude_only_accel': 399.12162055371647, 'min_extrude_temp': 170.0}}, 'save_config_pending': False}}

alfrix, 2022-03-02:
by default the screen is blanked until touched after a period determined in the settings, this is to prevent screen burning, again you can disable it in the settings
your log seems to be incomplete
you never specified what hw and workflow triggers this

dadonali, 2022-03-02:
Yes, unfortunately I can't attach the whole log here because it's too long.

I had the printer on longer now and the display was black again and could no longer be woken up by touching it.

The following was in the log after the blank display

--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.0-104-gc77bdfe
2022-03-02 13:07:16,239 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'EventButton' object has no attribute 'touch'
Traceback:   File "/home/pi/KlipperScreen/ks_includes/widgets/graph.py", line 43, in event_cb
    if ev.touch.type == Gdk.EventType.TOUCH_BEGIN:
NoneType: None
2022-03-02 13:07:17,295 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'EventButton' object has no attribute 'touch'
Traceback:   File "/home/pi/KlipperScreen/ks_includes/widgets/graph.py", line 43, in event_cb
    if ev.touch.type == Gdk.EventType.TOUCH_BEGIN:
NoneType: None
2022-03-02 13:07:17,505 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'EventButton' object has no attribute 'touch'
Traceback:   File "/home/pi/KlipperScreen/ks_includes/widgets/graph.py", line 43, in event_cb
    if ev.touch.type == Gdk.EventType.TOUCH_BEGIN:
NoneType: None
2022-03-02 13:07:18,683 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'EventButton' object has no attribute 'touch'
Traceback:   File "/home/pi/KlipperScreen/ks_includes/widgets/graph.py", line 43, in event_cb
    if ev.touch.type == Gdk.EventType.TOUCH_BEGIN:
NoneType: None
2022-03-02 13:07:18,915 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'EventButton' object has no attribute 'touch'
Traceback:   File "/home/pi/KlipperScreen/ks_includes/widgets/graph.py", line 43, in event_cb
    if ev.touch.type == Gdk.EventType.TOUCH_BEGIN:
NoneType: None

alfrix, 2022-03-02:
that is not a problem, it happens when you touch the graph, that's why i ask for the full log, zip it if necessary
and you haven't answered my question on what hw and workflow causes this behavior of not responding anymore

alfrix, 2022-03-05:
None of the logs are long enough to see the moment when screen blanking is turned on 
KlipperScreen2.log you set it to 3600, but the log is not long enough.

i see in the first attached log that you set screen blanking to off:
2022-03-01 20:29:09,109 [screen.py:set_screenblanking_timeout()] - Changing power save to: off
does this happen anyway after a period?  because it shouldn't and this will indicate a problem outside of KS

you need to extract the full log when it happens (do not reboot), the Xorg log may also bring some light

with workflow i mean what you configure and do to try to replicate

dadonali, 2022-03-07:
yeah nice,  i had changed the screensaver time to 5mins and now everything is fine, no blackout until now and wake up works fine

closed 2022-03-07 by alfrix