KlipperScreen/KlipperScreen #538: Black blank screen and high CPU usage

KSchimmel1992, 2022-03-09:
Hi Guys, 
I actually have installed v0.2.0-105 on a Raspi 3B with a Waveshare 5" Screen and it works very good. Thanks for your work!

But if i update to the latest version, the screen shows only the bootup lines and the splashscreen doesn't appear - it stays black. 
Also my CPU usage goes in about 5 minutes to 99 % and the Raspi doesn't react. 

I hope someone can help. 

Thanks a lot. 
Kevin (Germany - not native English speaker)

Eifel-Joe, 2022-03-09:
Hi Kevin,
same here. Worked fine with 0.2.0-123. Since update to 0.2.0-125 KlipperScreen seems to start but screen is blank. No touch to wake up. Using Pi4 and BTT PITFT50. CPU usage at 33 to 50% without doing anything on the pi. Normally it stays way under 10% in this state.

alfrix, 2022-03-09:
without log files i cannot provide any support, cpu usage maybe high because ks is bootlooping from your description, but again no logfile...

Eifel-Joe, 2022-03-09:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8218703/KlipperScreen.log)

Here is my Klipperscreen.log. I think you are right with bootlooping. If i restart KS the screen turns to console and gets blank again. So far normal behavour. But it stays blank.

alfrix, 2022-03-09:
it seems to be bootlooping until it runs out of memory, but no apparent reason from the logs

maybe /var/log/Xorg.0.log may have a clue if it's a driver related thing

can you test adding this to klipperscreen.conf
```
[main]
use_dpms: False
```

Eifel-Joe, 2022-03-09:
I had a look by myself in the log-file and in the spash_screen.py which was mentioned in the last lines in the log. In spash_screen.py the power device is named "printer". My power device was called "Printer". I changed it to "printer" and everything is working again.
I suppose that i have to ad "power_devices: Printer" in KlipperScreen.conf. Is that right?

alfrix, 2022-03-09:
ah ok, i was suspecting from the new screensaver function, but nevermind

it comes from the configuration of the power_devices, (also very recent)
the current code should add a button if 
 if device == printer: (your case since you didn't rename the printer and use the power device named Printer) 
and if device in power_devices: (the new configurable list
if you add it to klipperscreen.conf it should be under the printer section (it's case sensitive)
```
[printer Voron]
moonraker_host: 127.0.0.1
power_devices: printer
```

i will investigate this more deeply tomorrow

Eifel-Joe, 2022-03-09:
I set this option now in my other three printers which were still running 123 and updated them to 125. Everything is fine and is working as expected.

ReXT3D, 2022-03-10:
FWIW, I believe that I am seeing the exact same issue on two different installations with 0.2.0-125.  One is running Fluiddpi on Pi 3 and the second one is running Mainsailos on Pi 4.  They both are very stubborn to wake up taking multiple touches and a long time.  The backlight comes on on the first touch though.  Both screens are Waveshare 4.3 DSI.

Thank you for this excellent software.

KSchimmel1992, 2022-03-10:
Thanks a lot! 
It works really fine now!

closed 2022-03-10 by alfrix

alfrix, 2022-03-10:
it was an issue with the default configuration, it should be fixed now, thank you for your help

alfrix, 2022-03-10:
previous solution wasn't good enough, i found it still misbehaved under some circumstances where the user copied the previous default.conf, for that reason using the power_device option in [printer name] is now mandatory for this button to appear

ReXT3D, 2022-03-28:
I am not sure if this is still the same issue, but I experienced a few more events where touching the screen did not bring the display content back.  The backlight "woke up" and the screen was recognizing the touch input (I accidentally started the parts fan), but the display was all black.

Is there anything I can do to help identify the root cause?  Log collection, etc?

Thank you.

alfrix, 2022-03-30:
probably not the same issue unless you haven't updated, grab the log just after it happens

ReXT3D, 2022-04-03:
> probably not the same issue unless you haven't updated, grab the log just after it happens

Thanks for the suggestion.  I was about to reply that, immediately after I wrote my comment down, the anomalous behaviour disappeared.  However, it just happened again.  It is very intermittent and seems to happen once every 10-20 wake-up events.  The log file is attached, but I am not sure if there is anything helpful in it.

[KlipperScreen.log.zip](https://github.com/jordanruthe/KlipperScreen/files/8405107/KlipperScreen.log.zip)

I was able to get the display back after several additional finger presses on the black screen.  Once again, this is 4.3" Waveshare DSI IPS screen that has been working beautifully for months until the more recent Klipperscreen builds.

ReXT3D, 2022-11-10:
FWIW - I continued to experience issues with the display not showing content following a wake-up until one of the recent commits.  I have not seen any such issues since.

commit 65248bfe (2022-03-10): config: fix malfomed default config fixes #538