KlipperScreen/KlipperScreen #546: Control LED like a fan

Tim2309, 2022-03-15:
Hello,

is there a way to control a LED like the cooling fan or nevermore?
Only way which work is to set the LED as a fan in the config but that cause wrong icons for the LEDs.

alfrix, 2022-03-27:
is it configured as an output_pin?

Tim2309, 2022-03-27:
Yes, the config is:

[output_pin caselight]
pin: PB5
pwm: true
value: 0.3
shutdown_value: 1.0
cycle_time: 0.001

yzheka, 2022-04-08:
@Tim2309 But what if leds can be only on/off, no pwm?

Tim2309, 2022-04-08:
I think most people use led strips and they are always dimable

yzheka, 2022-04-08:
My not

Tim2309, 2022-04-08:
A simple on/off is already integrated, but I don’t know how to use it.

bartlammers, 2022-04-19:
> Right. On/off is working on klipper, but it is not available via klipperscreen. The only workaround is toggle macro

In your KlipperScreen.conf :
`[menu __main actions lights]
name: {{ gettext('Lights') }}
icon: light
method: printer.gcode.script
params: {"script":"_TOGGLE_CASELIGHT"}`

In your printer.cfg :
`[gcode_macro _TOGGLE_CASELIGHT]
gcode:
  SET_PIN PIN=caselight VALUE={(not printer['output_pin caselight'].value)|int}`

This works for me, but only as an on/off toggle indeed

commit 25e6110d (2022-08-01): output_pin panel close #546
closed 2022-08-01 by alfrix