KlipperScreen/KlipperScreen #547: Raspberry Pi 7" Touchscreen Display // Touch not working

nshoobridge, 2022-03-15:
Display is showing screen normally. Touch function is not working. Raspberry 4 with Pi 7" Touchscreen Display.

hawkeyexp, 2022-03-15:
i2c enabled ? The touch control and brightness is released this way.

nshoobridge, 2022-03-16:
SDA and SCL Pins are currently not connected. Thought that is is not needed on PI4. Will try and provide feedback. Thanks

hawkeyexp, 2022-03-16:
Wiring is not needed - it is also connected by the dsi cable.
Check your config.txt if the i2c interface is eabled or use raspi-config to enable i2c interface

nshoobridge, 2022-03-20:
Any other things where i should look into?

hawkeyexp, 2022-03-20:
Not really - if possible do a plain test - use a clean a sdcard and check it with a fresh installed raspbian os (full) without any mods if hardware itselve is ok

alfrix, 2022-03-27:
you will have to test your hw and confirm it's working with some other distro as hawkeyexp, and then try to install it on raspbian lite or any os you like
but since this is not really a KS issue i will close this

closed 2022-03-27 by alfrix