KlipperScreen/KlipperScreen #548: Bug: Power device detection fails after commits for mandatory... :-(

hawkeyexp, 2022-03-15:
[image]

Before the changes i got the power on button on splash ...

Also tried manually add the power device listing in config but same...

alfrix, 2022-03-15:
it's probably your config that is wrong, since you did not paste the full log you need to at least show the config file

hawkeyexp, 2022-03-16:
You can see in the screenshot the power devices are detected and the array is listing three devices including a power device with same name as the printer itselve.
It is also visible that klipperscreen can't match printer name with power device name.

The system is fully working and the power button was shown before the change.

hawkeyexp, 2022-03-16:
Possible it is a problem woth multi printer entries ?

[printer VORON24]
moonraker_host: 127.0.0.1
moonraker_port: 7125
language: de
show_cursor: False

[printer AI3MS]
moonraker_host: 192.168.188.210
moonraker_port: 7125
language: de
show_cursor: False

alfrix, 2022-03-16:
malformed config, ks configuration parsing needs to be a lot better, to warn users, but you should look at the docs once more
https://klipperscreen.readthedocs.io/en/latest/Configuration/

```
[main]
language: de
show_cursor: False

[printer VORON24]
# This power device is the 'associated power device' that will turn on in the splash screen
power_devices: VORON24

[printer AI3MS]
moonraker_host: 192.168.188.210
# It can be a list
power_devices: Trockenbox, Haupschalter
```

alfrix, 2022-03-21:
There was an issue affecting this, should be fixed with 1dee91456493ea659f2ee637c57450141a0278c2

hawkeyexp, 2022-03-22:
Something is going really wrong now - before you posted i manged it by correctly adding to each printer the power device config but after your fix power button is shown but not longer working :-(

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8328073/KlipperScreen.log)

alfrix, 2022-03-24:
i can see that the button is detected
Connecting to VORON24
VORON24 associated power devices: ['VORON24']
Power devices: ['Hauptschalter', 'VORON24', 'Trockenbox']
Found ['VORON24'], Adding power button

but never pressed, otherwise it will show sending power on

hawkeyexp, 2022-03-25:
it is pressed and notify message shows up on top of screen(green with correct device) but it seems command is not send - no entry in KlipperScreen log as you noticed and also no info in moonraker log :-( But the funny thing - if device is manually powered on and klipper is running it is possible to switch all power devices without any problem so setup must be correct.
If power device VORON24 is off display shows moonraker connected, klipper disconnected and the retry counter is counting ...

alfrix, 2022-03-25:
from my testing it seems that moonraker only responds to power on if klipper is connected to moonraker

hawkeyexp, 2022-03-25:
I will do some more checks this evening including reverting to older git revision where it was working to identify the breaking point.

hawkeyexp, 2022-03-26:
I switched back until commit 7dc74f74f48116a662cdd32cde02a205b50136df from march 18 where power button works without any problems from splash screen while klipper is offline and only moonraker is running. Switching forward to march 19 and the fun begins :-)

alfrix, 2022-03-27:
i did some changes that may affect this, can you test 2.1?

cd KlipperScreen
git checkout 2.1

it's a bit weird that no one else is reporting an issue with the button

hawkeyexp, 2022-03-27:
Hi, i have switched to 2.1 but now powert button is no longer visible :-(
Then a opened menu and entered power devices and toggled the VORON24 device but no action - only a log entry. During printer powered on i can power off it this way - crazy ...
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8358494/KlipperScreen.log)

And about reports - possible not in use by many users - don't know

alfrix, 2022-03-27:
i see that from the power panel it did what it supposed to do:

```
2022-03-27 22:59:44,463 [screen.py:_load_panel()] - Loading panel: power
...
2022-03-27 22:59:45,692 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-03-27 22:59:45,706 [power.py:on_switch()] - Power toggled VORON24
2022-03-27 22:59:45,708 [KlippyWebsocket.py:power_device_on()] - Sending machine.device_power.on VORON24
```
to be clear, from the web interface it does turn on but from ks doesn't

i believe the issue came from the memory leak fixing, where printer data is reset on reconnect, maybe i was too aggressive, i will investigate this further, and prepare a bugfix branch for you to test

if i'm on the right track 4899701ae356ff67b5521b40d0b7ff797562c05b is the last commit that should work

git reset --hard 4899701ae356ff67b5521b40d0b7ff797562c05b

hawkeyexp, 2022-03-28:
yes that's correct - control via mainsail without any problems.

the track is not fully correct - with this commit i can power of but not longer power on (tested in power menu cause power button in splash not shown)

hawkeyexp, 2022-03-29:
Just a notice: with commit 6f86448 power button is not visible and not working in power menu but if i enable the power device the power button is shown during klipper startup on splash - funny :-)

hawkeyexp, 2022-03-30:
hm crazy stuff - my power device is a tasmota device but it should make no difference for klipperscreen - hm have you also tried it with a second printer in the config like it is in my config ?

hawkeyexp, 2022-03-30:
I will do one thing:
full clean setup on a second sdcard to exclude possible old stuff to check if i can reproduce it - possible on the wekend.

commit 6f864484 (2022-03-28): Fix issues while Moonraker is connected but klipper is disconnected

fixes #548
closed 2022-04-01 by alfrix