KlipperScreen/KlipperScreen #551: Raspberry Pi 7 "TFT display (800x480)

AntoszHUN, 2022-03-18:
There have been several issues with the display since the last update.
  - One is that the display shows vertical streaks on the display after waking up from sleep. It's like the resolution isn't good. I checked the Display setup with the (sudo raspi-config) command but can't find the resolution of my display. Could this be a Sudo side mistake?
  - The other error I noticed is that the Raspberry processor and the MCU temperature display of the connected motherboard have disappeared from the display.

Thank you in advance for your help!

alfrix, 2022-03-19:
_he display shows vertical streaks on the display after waking up from sleep.

sounds like a hardware configuration or hardware quirk

_Raspberry processor and the MCU temperature display of the connected motherboard have disappeared from the display._

nothing was changed to affect this

Since there was no log attached, i cannot help you any further

AntoszHUN, 2022-03-19:
> 

> _he display shows vertical streaks on the display after waking up from sleep.
>
> sounds like a hardware configuration or hardware quirk
>
> _Raspberry processor and the MCU temperature display of the [...]

Since then, I’ve gotten so far that I checked the Raspi configuration with the sudo raspi-config command. The display settings do not include the resolution. (800x480) this is the resolution of the factory Raspberry Pi 7 "TFT Touch Display. I attached the picture.
[image]

alfrix, 2022-03-19:
you will have to set it on /boot/config.txt
framebuffer_width=800
framebuffer_height=480

i don't know about fluidd, but KS will only show temps configured in klipper

printer.cfg
```
[temperature_sensor MCU]
sensor_type: temperature_mcu
min_temp: 0
max_temp: 85

[temperature_sensor Pi]
sensor_type: temperature_host
min_temp: 0
max_temp: 85
```

i can confirm they work as expected

AntoszHUN, 2022-03-19:
> 

> you will have to set it on /boot/config.txt (needs a reboot) framebuffer_width=800 framebuffer_height=480
> 
> i don't know about fluidd, but KS will only show temps configured in klipper
> 
> printer.cfg
> 
> ```
> [temperature_sensor MCU]
> sensor_type: temperature_mcu
> min_temp: 0
> max_temp: 85
> 
> [temperature_sensor Pi]
> sensor_type: temperature_host
> min_temp: 0
> max_temp: 85
> ```
> 
> i can confirm they work as expected

Yes. You have these settings. However, I also set the resolution myself in config.txt. Result: no image display after boot. Only a black backlit image that is vertically striped. I note that the display works flawlessly during boot.

alfrix, 2022-03-27:
pi and mcu work as expected and you have shown in a recent issue that is working for you too
for the hardware issue follow the manufacturer instructions and ask in discord, maybe someone with your hw can point something out

closed 2022-03-27 by alfrix