KlipperScreen/KlipperScreen #552: Power button disappeared from splash screen.

LazaroFilm, 2022-03-19:
The power button disappeared from the splash screen next to restart /firmware restart etc... I can still power the printer from the power menu but it's a handful of extra taps. [image]

alfrix, 2022-03-19:
https://klipperscreen.readthedocs.io/en/latest/Configuration/#printer-options
you need to add the moonraker power_device (or devices) that turn on the printer
example of klipperscreen.conf
```
[main]
show_cursor: False

[printer Voron 2.4]
# This power device is the 'associated power device' that will turn on the printer in the splash screen
power_devices: PSU
```

closed 2022-03-19 by alfrix

alfrix, 2022-03-21:
There was an issue affecting this, should be fixed with 1dee91456493ea659f2ee637c57450141a0278c2