KlipperScreen/KlipperScreen #553: Klipperscreen showing full screen colors

Rasmusbdk, 2022-03-19:
At some point in each print, my klipperscreen starts showing full screen colors. Blue, red, green, white ish. I thought it was a cable connection problem, but it goes away as soon as the print is done. Every time. 

Any ideas? 

https://user-images.githubusercontent.com/32739961/159134555-d4c9cb3f-f616-461c-b2a1-6c9fbb02bfb9.mp4

Thanks in advance. And Thanks for the great project!

alfrix, 2022-03-19:
Probably dpms, try turning it off

Rasmusbdk, 2022-03-19:
Thanks. I am not familiar with dpms, and Google didnt make me any less confused. Can you give me a hint on how to turn it off?

Thanks

Rasmusbdk, 2022-03-19:
Looking at the colors, I was  sure that something has gone wrong.. But turns out its actually a screensaver - when I touch the screen, they dissappear and the menu is shown again. Any idea how to turn off this confusing screensaver? Thanks

alfrix, 2022-03-19:
[image]

closed 2022-03-27 by alfrix

Rasmusbdk, 2022-03-27:
Turns out it is Actually a screensaver. A touch on the screen makes it go away.

Disabling dpms doesnt help though.

Can I do something else to make it just turn off the screen instead?

alfrix, 2022-03-30:
instead of improving, this screens are getting worse, manufacturers should be ashamed of the crap they are making, don't worry i have a plan b, i will reuse the internal screensaver that i created for dpms for the no dpms mode