KlipperScreen/KlipperScreen #556: [BUG] Layout not alligned with latest commits from 22 march | Not configured entries in menu

hawkeyexp, 2022-03-24:
After updating to latest commits the layout has bugs in two screens (homing and config always in 3rd row):

[image]
[image]

Additional quad gantry level is shown in homing where it is not configured (seems to come from default config...)

alfrix, 2022-03-24:
never post bugs without a log file, from the logfile you provided in the other issue i can see that is your config that is wrong

you defined in your config:
```
[menu __main config quad_gantry_level]
name = {{ gettext('Quad Gantry Level') }}
icon = home-z
method = printer.gcode.script
params = {"script":"QUAD_GANTRY_LEVEL"}
enable = {{ printer.quad_gantry_level is defined }}
```
which is not part of the default entries...

please remove all the non custom entries from your configuration, the merge of config should be working now

the quad gantry level is in your homing menu, because the printer has it defined, if you want to replace it with your entry then you need to properly replace it changing the script
```
[menu __main homing quad_gantry_level]
name: Custom Quad Gantry Level
icon: z-tilt
method: printer.gcode.script
params: {"script":"my_custom_macro"}
```
notice that ```[menu __main homing quad_gantry_level]``` redefines the default, the rest is custom

if you don't like this and want a full set of custom menus use the proper config setting as described in the docs

also language should be on [main] not under [printer]

alfrix, 2022-03-24:
and for the getting out of screen, is kind of unavoidable, i will at some point make a much more strict config parser that will not load until your config is properly set

hawkeyexp, 2022-03-24:
don't understand me wrong but it is a bit frustrating if one thing was working for weeks and on day x the word changes completly...

for example the qgl is one thing i won't have inside homing so i removed it in the past and placed in in settings and now like jack in a box it is back without any config change :-)

hawkeyexp, 2022-03-24:
And about the layout - what is wrong with it - to many entries etc. or what is the point in this case ? In other screen three rows works well...

alfrix, 2022-03-24:
Merging default menus is what most people wanted, that's why I fixed it and changed to this behavior, if you want the old way, turn off default menus as said.

3 rows doesn't work if the labels are too long, I think I may have an idea on how to handle it, I will queue it for 2.1

closed 2022-03-25 by hawkeyexp