KlipperScreen/KlipperScreen #561: Display error

AntoszHUN, 2022-03-27:
The error persists on the display. I don't know where to write. The display will be streaked if I wake up. Sometimes the image flickers as you live. Resolution cannot be selected after using the raspi-config command. The resolutions do not include the Raspberry Pi 7 "TFT resolution which would be 800x480. FluiddPi is installed. There was nothing wrong with Mainsail, it worked flawlessly. Who should I write to?
[image]

alfrix, 2022-03-27:
it's very hard to understand you, from what i can gather it seems like a hardware related issue but, you say that it worked with mainsail ? maybe you mean mainsailos vs fluiddpi and it's a distro related thing?, idk ask in discord, this is not an issue with KS

closed 2022-03-27 by alfrix

AntoszHUN, 2022-03-27:
> 

I understand. All right!