KlipperScreen/KlipperScreen #564: Can't shutdown or reboot the Rapsberry Pi via Klipperscreen.

JWRu, 2022-03-29:
I have a problem with KlipperScreen since an update during the last days:
When I select Menu -> System -> System Shutdown -> I have two choices: "Continue" and "Cancel".  Both fields bring me back to the "System" menu - shutdown is not possible. The same happens with "System Restart.

JWRu, 2022-03-29:
Update. It works when the printer is powered on, but it doesn't when the printer is powered off.

alfrix, 2022-03-29:
this is most likely fixed in the 2.1 branch, you can test it if you want 
in ks folder:
```
git fetch
git checkout 2.1
```

restart klipper screen

JWRu, 2022-03-30:
It works with branch 2.1

JWRu, 2022-03-31:
When will the master branch be updated to fix this issue?

alfrix, 2022-04-01:
done

closed 2022-04-01 by alfrix