KlipperScreen/KlipperScreen #568: Add Layer Info as a First Class Information Line

adawalli, 2022-04-01:
While the M117 method described in the docs is an interesting workaround, it doesn't come for "free".

For example, if you also use the Telegram Plugin, you will get a telegram notification for every single layer (for every M117 message).

Would it be possible to add proper first-class support for this important piece of information to the primary home panel?

closed 2022-05-01 by alfrix

svcsectool, 2022-05-01:
I see my MR was closed but I actually don't see the layer count on the new UI - am I missing something?

adawalli, 2022-05-01:
I see my MR was closed, however, I don't see layer information in the new UI - where can I find it?

alfrix, 2022-05-02:
click/press the Z position
all the information in the new main print status are buttons that take you elsewhere

adawalli, 2022-05-02:
oh wow, that's amazing. thanks! looks great!

PR #592 (2022-05-01): Improve Job status
Adds heater_generic and/or temp_sensors besides extruder/bed (because of the limited space maybe only 1)
Temps are now buttons and act as a shortcut to the temp panel
Adds fan_generic to the fan label and it's now a shortcut to the fan panel
The colors and size of the progress circle were changed to be more subtle.
Margins between items were augmented, because they are buttons now
Adds 3 information pages: move, extrusion and time:
* move can be opened with speed or z buttons
* time with elapsed/remaining buttons
* extrusion with the extrude_factor / flowrate button

TODO
-  [... cut]