KlipperScreen/KlipperScreen #570: screen blur

AntoszHUN, 2022-04-03:
It was corrupted with the last update. The display does not turn off only the background color is visible. The controls disappear. But if I tap on him, he'll come back. Darkening does not work.

[image]

alfrix, 2022-04-03:
that is expected is DPMS is off, provide a logfile if that isn't the case

AntoszHUN, 2022-04-03:
> that is expected is DPMS is off, provide a logfile if that isn't the case

DPMS is not turned off because you don't even know what it is.

AntoszHUN, 2022-04-03:
I tried again. I turned off DPMS and it does the same. It doesn't darken.

AntoszHUN, 2022-04-23:
Will there be a solution to this? The display still does not dim. Only the background color is visible.

alfrix, 2022-04-23:
> I turned off DPMS and it does the same. It doesn't darken.

without DPMS it will not turn off

with DPMS it should, i will take a look using an HDMI tv to check if it's working

AntoszHUN, 2022-04-23:
> 

> > I turned off DPMS and it does the same. It doesn't darken.
> 
> without DPMS it will not turn off
> 
> with DPMS it should, i will take a look using an HDMI tv to check if it's working

DPMS is on for me. As you can see in the picture.

[image]

alfrix, 2022-06-19:
i have tested dpms and it works as expected, so i dont think this is an issue with KS
you should ssh to the pi and enter this command to test your screen
```
DISPLAY=:0 xset dpms force off
```
if the screen turns off with the command then reopen this issue and provide KlipperScreen.conf
if the screen doesn't turn off then its an issue with your hardware or installation that doesn't work with DPMS

closed 2022-06-19 by alfrix

AntoszHUN, 2022-06-19:
> i have tested dpms and it works as expected, so i dont think this is an issue with KS you should ssh to the pi and enter this command to test your screen
>
> ```
> DISPLAY=:0 xset dpms force off
> ```
>
> if [...]

I gave the command via PUTTY. I got this in response and the display did not go dark.
[image]

alfrix, 2022-06-19:
interesting, try the following command and tell me if it works:
```
DISPLAY=:0 xset s blank
```
then test again with this:

```
DISPLAY=:0 xset dpms force off
```