KlipperScreen/KlipperScreen #575: unable to use z_calibrate on delta..

danorder, 2022-04-05:
bug delta_calibrate method = manual doesn't use a defined probe so the menus are now disabled with recent commits. 
follow up request. ability to use delta_calibrate_manual instead of default zcalibrate "  eg run though the 7 points using the same menu.  
it could be macroed prior.  

example 
[gcode_macro Z_ENDSTOP_CALIBRATE] #klipperscreen workaround to run delta calibrate on klipper screen. 
#rename_existing: DELTA_CALIBRATE_MANUAL 
description: run klipper screen zoffset menu as delta calibrate manual instead. 
gcode: 
  G28
  DELTA_CALIBRATE METHOD=manual

lower nozzle using buttons , hit accept button  it moves to the next point etc ,  accept button  , save

alfrix, 2022-04-05:
why method=manual and not automatic probe?

closed 2022-04-05 by alfrix

danorder, 2022-04-08:
> Closed #575 <https://github.com/jordanruthe/KlipperScreen/issues/575> via
> 4bace34

because not all deltas have a probe and, not all probes are accurate
enough. as listed in klipper docs. the recommended leveling procedure is
manual.

commit 4bace348 (2022-04-05): zcalibrate: add manual delta_calibrate fix #575