KlipperScreen/KlipperScreen #576: feed rate / extrusion buttons not working.

danorder, 2022-04-05:
bug: during a print neither button do anything after new updates.

alfrix, 2022-04-05:
just tested, both work as expected, will need a logfile containing the moment that this happened

danorder, 2022-04-06:
sure

danorder, 2022-04-06:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8422707/KlipperScreen.log)

danorder, 2022-04-06:
i couldn't say what commit that started on.  i was 10-12 or so behind.

alfrix, 2022-04-06:
nothing in the logs that indicates an error with the buttons

danorder, 2022-04-06:
i didn't see anything either. when i hit it i don't' see any output etc.   i'll try reinstalling

danorder, 2022-04-07:
nope issue remains the same.  rolled back 10 days back it works as normal. so im not sure what the issue is.  their is another post with it failing to start i had that as well.

danorder, 2022-04-12:
got it to spit out the log for the buttons as well 
2022-04-12 02:01:18,584 [printer.py:reinit()] - Klipper version: v0.10.0-368-g4a8a76ea
2022-04-12 02:01:18,585 [printer.py:reinit()] - # Toolcount: 1
2022-04-12 02:01:18,585 [printer.py:reinit()] - # Extruders: 1
2022-04-12 02:01:18,585 [printer.py:reinit()] - # Temperature devices: 3
2022-04-12 02:01:18,585 [printer.py:reinit()] - # Fans: 3
2022-04-12 02:01:18,585 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe: {'objects': {'bed_mesh': ['profile_name', 'mesh_max', 'mesh_min', 'probed_matrix'], 'configfile': ['config'], 'display_status': ['progress', 'message'], 'fan': ['speed'], 'gcode_move': ['extrude_factor', 'gcode_position', 'homing_origin', 'speed_factor'], 'idle_timeout': ['state'], 'pause_resume': ['is_paused'], 'print_stats': ['print_duration', 'total_duration', 'filament_used', 'filename', 'state', 'message'], 'toolhead': ['homed_axes', 'estimated_print_time', 'print_time', 'position', 'extruder', 'max_accel', 'max_accel_to_decel', 'max_velocity', 'square_corner_velocity'], 'virtual_sdcard': ['file_position', 'is_active', 'progress'], 'webhooks': ['state', 'state_message'], 'extruder': ['target', 'temperature', 'pressure_advance', 'smooth_time'], 'heater_bed': ['target', 'temperature'], 'temperature_sensor mcu_temperature': ['target', 'temperature'], 'temperature_sensor rpi_temperature': ['target', 'temperature'], 'heater_fan LED': ['speed'], 'heater_fan heatsink_fan': ['speed']}}
2022-04-12 02:01:18,586 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks&extruder&heater_bed&temperature_sensor mcu_temperature&temperature_sensor rpi_temperature&fan&heater_fan LED&heater_fan heatsink_fan
2022-04-12 02:01:18,866 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2022-04-12 02:01:18,903 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_sensor rpi_temperature', 'temperature_sensor mcu_temperature', 'extruder']
2022-04-12 02:01:18,903 [printer.py:change_state()] - Changing state from 'disconnected' to 'printing'
2022-04-12 02:01:18,903 [printer.py:change_state()] - Adding callback for state: printing
2022-04-12 02:01:18,904 [files.py:initialize()] - Gcodes path: /home/pi/gcode_files
2022-04-12 02:01:18,904 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2022-04-12 02:01:18,905 [screen.py:init_printer()] - Printer initialized
2022-04-12 02:01:18,906 [screen.py:_load_panel()] - Loading panel: job_status
2022-04-12 02:01:18,906 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/job_status.py
2022-04-12 02:01:18,994 [screen.py:wake_screen()] - Screen wake up
2022-04-12 02:01:19,035 [job_status.py:set_state()] - Changing job_status state from 'standby' to 'printing'
2022-04-12 02:01:19,037 [job_status.py:update_file_metadata()] - Cannot find file metadata. Listening for updated metadata
2022-04-12 02:01:19,038 [screen.py:show_panel()] - Attaching panel job_status
2022-04-12 02:01:19,042 [screen.py:show_panel()] - Showing back. count: 0
2022-04-12 02:01:19,053 [screen.py:show_panel()] - Current panel hierarchy: ['job_status']
2022-04-12 02:01:19,514 [job_status.py:update_file_metadata()] - Update Metadata. File: dog_food_scoop.gcode Size: 22578352
2022-04-12 02:01:53,621 [screen.py:_load_panel()] - Loading panel: fine_tune
2022-04-12 02:01:53,621 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/fine_tune.py
2022-04-12 02:01:53,640 [fine_tune.py:initialize()] - FineTunePanel
2022-04-12 02:01:53,652 [fine_tune.py:change_percent_delta()] - ### Delta 1
2022-04-12 02:01:53,699 [screen.py:show_panel()] - Attaching panel job_status_fine_tune
2022-04-12 02:01:53,702 [screen.py:show_panel()] - Showing back. count: 1
2022-04-12 02:01:53,712 [screen.py:show_panel()] - Current panel hierarchy: ['job_status', 'job_status_fine_tune']
2022-04-12 02:01:56,148 [fine_tune.py:change_percent_delta()] - ### Delta 25
2022-04-12 02:01:56,149 [fine_tune.py:change_percent_delta()] - ### Delta 1
2022-04-12 02:01:56,150 [fine_tune.py:change_percent_delta()] - ### Delta 25

danorder, 2022-04-12:
> Closed #576 <https://github.com/jordanruthe/KlipperScreen/issues/576> via
> 8ad21ab

tested and confirmed working. thank you.
\