KlipperScreen/KlipperScreen #577: Software update not possible?

komandrik, 2022-04-06:
Today I'm trying to upgrade from version 0.2.1-5 to 0.2.1-14 via mainseyl, KIAUX, terminal console and everywhere I get a refusal to update
[image]

[image]

Can I find out what has changed and how to fix it?

alfrix, 2022-04-06:
my fault, sorry, i don-t understand why moonraker isn't marking the repo as dirty as it should

i think the easiest way will be
delete the contents of KlipperScreen folder
click on up refresh in the webui, the repo will be marked as dirty
click reset soft or hard

manually in via ssh
cd KlipperScreen
rm -rf *
git fetch
git reset --hard @

axfb112, 2022-04-06:
> my fault, sorry, i don-t understand why moonraker isn't marking the repo as dirty as it should
>
> i think the easiest way will be delete the contents of KlipperScreen folder click on up refresh in the [...]
just tried but not vorking . same error

alfrix, 2022-04-06:
weid, that should work

a couple more options
delete the entire folder
and do
git clone https://github.com/jordanruthe/KlipperScreen.git

uninstall and then reinstall via KIAUH

closed 2022-04-06 by komandrik

miklschmidt, 2022-04-06:
@alfrix Someone forced pushed to master?

alfrix, 2022-04-06:
it was me wrote master instead of dev, never force pushing again on anything it's just too dangerous, but now it's too late

miklschmidt, 2022-04-06:
@alfrix You can disable force pushing to master in the repository settings on github, then it won't happen in the future :)