KlipperScreen/KlipperScreen #580: klipper screen not loading after install.

Giibnn, 2022-04-07:
i have a pi 3b+ and BTT TFT50 2.0 screen. instructions on webpage and klipperscreen wont load.

Apr 07 03:33:43 fluiddpi xinit[1576]: Current Operating System: Linux fluiddpi 5.15.30-v7+ #1536 SMP Mon Mar 28 13:43:34 BST 2022 armv7l
Apr 07 03:33:43 fluiddpi xinit[1576]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  >
Apr 07 03:33:43 fluiddpi xinit[1576]: Build Date: 04 January 2022  10:24:42AM
Apr 07 03:33:43 fluiddpi xinit[1576]: xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support)
Apr 07 03:33:43 fluiddpi xinit[1576]: Current version of pixman: 0.40.0
Apr 07 03:33:43 fluiddpi xinit[1576]:         Before reporting problems, check http://wiki.x.org
Apr 07 03:33:43 fluiddpi xinit[1576]:         to make sure that you have the latest version.
Apr 07 03:33:43 fluiddpi xinit[1576]: Markers: (--) probed, (**) from config file, (==) default setting,
Apr 07 03:33:43 fluiddpi xinit[1576]:         (++) from command line, (!!) notice, (II) informational,
Apr 07 03:33:43 fluiddpi xinit[1576]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Apr 07 03:33:43 fluiddpi xinit[1576]: (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Thu Apr  7 03:33:43 2022
Apr 07 03:33:43 fluiddpi xinit[1576]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE)
Apr 07 03:33:43 fluiddpi xinit[1576]: Fatal server error:
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE)
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE)
Apr 07 03:33:43 fluiddpi xinit[1576]: Please consult the The X.Org Foundation support
Apr 07 03:33:43 fluiddpi xinit[1576]:          at http://wiki.x.org
Apr 07 03:33:43 fluiddpi xinit[1576]:  for help.
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE)
Apr 07 03:33:43 fluiddpi xinit[1576]: (EE) Server terminated with error (1). Closing log file.

Giibnn, 2022-04-07:
here is my Xorg log

28.704]    compiled for 1.20.11, module version = 1.20.11
[    28.704]    Module class: X.Org Video Driver
[    28.704]    ABI class: X.Org Video Driver, version 24.1
[    28.704] (II) LoadModule: "fbdev"
[    28.704] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    28.705] (II) Module fbdev: vendor="X.Org Foundation"
[    28.705]    compiled for 1.20.3, module version = 0.5.0
[    28.705]    Module class: X.Org Video Driver
[    28.705]    ABI class: X.Org Video Driver, version 24.0
[    28.705] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    28.705] (II) FBDEV: driver for framebuffer: fbdev
[    28.705] (EE)
Fatal server error:
[    28.705] (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission de>
[    28.705] (EE)
[    28.705] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org

Giibnn, 2022-04-07:
I have fixed the issue. I edited the Xwrapper.config file and added the line
'needs_root_rights=yes'

alfrix, 2022-04-07:
this is an OS thing that shouldn't be needed
but i will add it to the corresponding section in troubleshooting as one of the possible fixes
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#cannot-open-virtual-console

Giibnn, 2022-04-07:
Ok , what would you have recommended me to do to try and fix this issue? . Thanks

commit 0fb20466 (2022-04-08): docs: add another fix to Cannot open virtual console close #580
closed 2022-04-08 by alfrix