KlipperScreen/KlipperScreen #581: Bed Leveling Panel

Cyypher, 2022-04-07:
When I go to Configuration > Bed Level. In the documentation for Klipper it says to set the [bed_screws] to the Nozzle over the Bed Screws. These are accurate and tested manually just to confirm.

[Klipper Docs](https://www.klipper3d.org/Config_Reference.html#bed_screws)

Attached is also all my Klipper Config Files, a Screenshot of the Screen in question, and the klipperscreen.log
[Files](https://drive.google.com/drive/folders/1FUVqu1T3oF81ajuLnLjje9Rl1N6biXgG?usp=sharing)

I checked bed_level.py and saw that if [screws_tilt_adjust] exists the bed level panel will use these coordinates over [bed_screws]. Which seems to be working properly any adjustments to the [bed_screws] section of the printer.cfg do not affect the coordinates listed on the Bed level Panel.  But I run into issues with the bed coordinates being way off. By the amount of my Probe X Offset.

Two things.
1. I had to add 27mm to the X on each of the coordinates in [screws_tilt_adjust] for my probe to be over the bed screw. But shouldn't this already be done by the probe offset so when I press "Screw Adjust", which seems to run SCREWS_TILT_CALCULATE. The probe already knows to offset by 27mm. It seems I had to manually add the 27mm to each X in the [screws_tilt_adjust]. This creates the problem in #2.

2. The coordinates that are listed on the screen in bed level also add an additional 27mm to the coordinates in [screws_tilt_adjust]. Showing 80,40 / 329,40 / 329,275 / 80,275. So when I press any corner. My nozzle is off X by 54mm and in the end when I press the any one of the right corners it goes of the bed and runs into end of the gantry. Because I had to set stepper_x to 330 just to accommodate the move. Which should only be set to 310 since the bed is only 310mm to the X.

Cyypher, 2022-04-07:
User on Reddit DopeBoogie was able to look at my printer.cfg and found a -negative value in my [bltouch] area on setting x_offset was causing the problem. After changing the offset to a positive value. Both the Coordinates were correct, when pressing each corner. The nozzle went to the bed screw and when pressing Screws Adjust. The probe, probed all 4 corners and the probe went to each of the appropriate bed screws.

alfrix, 2022-04-07:
offsets, work the same way for every firmware
```
 *     +-- BACK ---+
 *     |    [+]    |
 *   L |        1  | R <-- Example "1" (right+,  back+)
 *   E |  2        | I <-- Example "2" ( left-,  back+)
 *   F |[-]  N  [+]| G <-- Nozzle
 *   T |       3   | H <-- Example "3" (right+, front-)
 *     | 4         | T <-- Example "4" ( left-, front-)
 *     |    [-]    |
 *     O-- FRONT --+
 ```

Cyypher, 2022-04-07:
Yes it seems my  confusion was the settings seems to be where is the nozzle in relate to the probe. No where is the probe in relate to the nozzle. So its a +27 mm right... no a -27 to the probe left.

closed 2022-04-29 by alfrix