KlipperScreen/KlipperScreen #582: Time on screen is one hour off

TheLion, 2022-04-07:
Just noticed that the time displayed on the KlipperScreen is one hour off, it is one hour behind the actual time. 

The time and timezone on the RPi running Klipper, Moonraker and KlipperScreen is set correctly.

We did just switch from daylight savings on March 7, maybe it has something to do with that. 

Already restarted everything, to check if that would fix it, but it didn't. 

Where is KlipperScreen getting the time/timezonefrom?

yzheka, 2022-04-08:
set correct timezone via raspi-config

TheLion, 2022-04-08:
> set correct timezone via raspi-config

Damn. You are right. Thought I set that correctly when setting up the Pi months ago. Guess I didn't.

Sorry for the n00b question.

closed 2022-04-08 by TheLion